emuflight/EmuFlight #547: Black box Error opening log: No logs in this file could be parsed successfully

youssefbach, 2021-04-23:
issue opening log file with emuflight configurator 0.3.5 but no issue when i use betaflight configurator 10.7.0
Kakute f7 v1.5

# diff

# version
# EmuFlight / KAKUTEF7V15 (KTF7) 0.3.4 Apr 16 2021 / 18:56:35 (1014a9331) MSP API: 1.50

board_name 
manufacturer_id 

# name

# resources
resource MOTOR 1 B01
resource MOTOR 2 E11
resource MOTOR 3 B00
resource MOTOR 4 E09

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature ESC_SENSOR

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 8192 115200 57600 0 115200
serial 6 1024 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1300 2075 0 0
aux 1 13 0 1300 2100 0 0
aux 2 35 3 1300 1700 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set acc_calibration = -96,-50,-75
set mag_hardware = NONE
set rssi_channel = 16
set blackbox_p_ratio = 64
set ibata_scale = 192
set yaw_motors_reversed = ON
set sdcard_dma = ON
set osd_vbat_pos = 2372
set osd_rssi_pos = 2401
set osd_tim_2_pos = 2423
set osd_throttle_pos = 2340
set osd_current_pos = 2411
set osd_mah_drawn_pos = 2392
set osd_warnings_pos = 2250
set osd_avg_cell_voltage_pos = 2404
set vtx_band = 5
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5658

# profile
profile 0

set pid_at_min_throttle = OFF

# rateprofile
rateprofile 0

set rate_center_sensitivity = 80
set rate_end_sensitivity = 90
set rate_center_correction = 20
set rate_center_weight = 85
set rate_end_weight = 50

# 
[LOG00001.zip](https://github.com/emuflight/EmuFlight/files/6366612/LOG00001.zip)
[LOG00002.zip](https://github.com/emuflight/EmuFlight/files/6366613/LOG00002.zip)

nerdCopter, 2021-04-24:
Log 1 opens for me.
Log 2 reports `Log truncated, no data`.

Most of us use the pre-release `nightly` builds.
Stable release: https://github.com/betaflight/blackbox-log-viewer/releases
Pre-release test "nighties": https://github.com/betaflight/blackbox-log-viewer-nightlies/releases

nerdCopter, 2021-04-24:
i tried using PlasmaTree on log2, but it fails also.

youssefbach, 2021-04-24:
Thank you for response, log1 created with betaflight, log2 with emuflight. This to explain issue come from emu and not hardware  ....

nerdCopter, 2021-04-24:
Thank you @youssefbach .

@Kaiowarez , any idea why `KAKUTEF7V15` BBL fail?

youssefbach, 2021-04-24:
@nerdCopter i record new log fly with same parameters and with emuflight. flight time above 5 minutes.
not issue now with betaflight blackbox explorer, i can see something :) but i dont know if its correct or not!
[LOG00003.zip](https://github.com/emuflight/EmuFlight/files/6370693/LOG00003.zip)

nerdCopter, 2021-04-24:
That looks bad, as if it cannot record data fluently.

If it is onboard flash, please set your logging rate to 500hz and try again.
If it is SDCard, consider reformatting or replacing SDCard.

[image]

youssefbach, 2021-04-25:
Kingstone Sdcard 32Go class 10, same card used with log1 (betaflight ) .

nerdCopter, 2021-04-25:
try smaller 16GB if you have the opportunity.  Maybe firmware drivers are not up to date.

youssefbach, 2021-04-25:
Ok thank you , i'll wainting for update

youssefbach, 2021-04-27:
@nerdCopter hi, any good news ?

nerdCopter, 2021-04-27:
> try smaller 16GB if you have the opportunity. [...]
^^^^^^^^^
* we are not expecting to update firmware drivers for 0.X versions.
* i can ask around see if someone else records on `KAKUTEF7V15` without error.

advancedtacmachine, 2021-06-03:
Had same issue with kakute used Sanyo 16gig and logged

youssefbach, 2021-06-15:
> Had same issue with kakute used Sanyo 16gig and logged

I think its a firmware drivers issue. waiting firmware update

nerdCopter, 2021-06-15:
no efforts will be made to update drivers, but we accept Pull Requests.
EmuFlight 0.4.0 on timeline to release, then all dev efforts will be on EmuFlight 1.0.0 which re-bases from Betaflight 4.X and inherits working systems.  That said 1.0.0 will be a long time from release.

nerdCopter, 2026-05-10:
AI Generated [investigation]

Investigated against **0.4.3 master**. Closing as stale — no target definition defect found.

**Target definition is correct.** `KAKUTEF7V15` is a named alias within `KAKUTEF7HDV` (same target folder, same `target.h`). Pin assignments, SPI buses, DMA, and UART definitions match Betaflight's `KAKUTEF7HDV` unified config exactly.

**SD card DMA:** EmuFlight uses `DMA2_Stream5 / CH3` for SPI1 TX (the alternate mapping on STM32F745). Betaflight 4.5 moved to `SPI1_TX_DMA_OPT 1` → `DMA2_Stream3` via the unified target system. Both are valid; no conflict or regression introduced by the EmuFlight assignment.

**Most likely root cause:** 32 GB Class 10 SD card. Class 10 is rated for 10 MB/s _sequential_ throughput but random 512-byte block write IOPS on video-optimized cards can be extremely low. Blackbox writes are many small sequential blocks under a tight scheduler deadline — buffer overrun causes truncation, not a firmware driver bug.

**No changes to this code path** between 0.3.4 and 0.4.3.

**Recommendation if still encountered on 0.4.3:**
- Use a 16 GB (or smaller) high-endurance or A1-rated SD card
- Set `blackbox_p_ratio = 32` or logging rate ≤ 500 Hz
- Confirm `sdcard_dma = ON` in CLI

If reproducible on 0.4.3 with a known-good card at a low logging rate, please open a new issue with the exact firmware version, SD card model, and a minimal diff.

closed 2026-05-10 by nerdCopter