emuflight/EmuFlight #563: [target request] Skystars F4 F405

nerdCopter, 2021-05-12:
**Vendor/Model**
* `Skystars F4 F405`
* `Skystars F4 DJI Flight Controller`
* https://a.aliexpress.com/_mOIHxA9

**CLI dump or Unified Target**
* unified: https://github.com/betaflight/unified-targets/blob/master/configs/default/SKST-SKYSTARSF405.config

requested by @ Dewx

BeauBrewski, 2021-11-17:
Needs tested!
Check gyro orientation, motor arrangement, flash, osd, vtx control, everything else if possible please update here
PROPS OFF FOR SAFETY

[EmuFlight_0.4.0_SKYSTARSF405_Build_961e842f9.zip](https://github.com/emuflight/EmuFlight/files/7556658/EmuFlight_0.4.0_SKYSTARSF405_Build_961e842f9.zip)

BeauBrewski, 2022-05-21:
Needs tested!
Check gyro orientation, motor arrangement, flash, osd, vtx control, everything else if possible please update here
PROPS OFF FOR SAFETY

[EmuFlight_0.4.0_SKYSTARSF405_Build_a5a716f14.zip](https://github.com/emuflight/EmuFlight/files/8747735/EmuFlight_0.4.0_SKYSTARSF405_Build_a5a716f14.zip)

Edit: Fixed gyro code, working now.

PR #1037 (2024-04-19): [target] add SKYSTARSF405
resolves #563