emuflight/EmuFlight #564: [target request] FPVCYCLEF401

nerdCopter, 2021-05-12:
**Vendor/Model**
* `FPVCYCLEF401`
* `FPVCycle 1S AIO (no Rx) - 1S ONLY`
* https://fpvcycle.com/products/fpvcycle-1s-aio-with-tbs-crossfire-tbs-pro32-nano-vtx

**CLI dump or Unified Target**
* unified: https://github.com/betaflight/unified-targets/blob/master/configs/default/TURC-FPVCYCLEF401.config

requested by @ Silent

nerdCopter, 2021-06-05:
technically made, but not merged due incomplete PR.
https://github.com/emuflight/EmuFlight/pull/509

BeauBrewski, 2021-06-11:
Need tested? https://cdn.discordapp.com/attachments/691022009865338950/805548071076364338/EmuFlight_0.3.3_TURC_FPVCYCLEF401_Build_acaae1776.hex

nerdCopter, 2021-06-11:
https://cdn.discordapp.com/attachments/691022009865338950/850773007253504050/EmuFlight_0.3.4_TURC_FPVCYCLEF401_Build_1330fb967.hex

nerdCopter, 2021-09-14:
`TURC_FPVCYCLEF401` was merged awhile back for 0.4.0 release

closed 2021-09-14 by nerdCopter