emuflight/EmuFlight #580: MATEKF411RX resource ADC_Batt not working!

Nalkhashab, 2021-05-24:
Hello..

I updated my MATEKF411RX FC to Emuflight then i checked my battery is saying 0v when i plug the battery 

checked the resource on EmuFlight CLI it was ADC_BATT B00 and i changed it to A01 its still not working

it is working on BetaFlight as A01

her is my resources on Emu CLI:

# resource
resource BEEPER 1 C15
resource MOTOR 1 B08
resource MOTOR 2 A00
resource MOTOR 3 B10
resource MOTOR 4 B07
resource PPM 1 A03
resource PWM 1 A02
resource PWM 2 A09
resource PWM 3 A10
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource LED 1 C13
resource RX_BIND_PLUG 1 B02
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 B03
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 B04
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource ADC_BATT 1 B00
resource ADC_CURR 1 B01
resource OSD_CS 1 B12
resource SPI_PREINIT_IPU 1 A04
resource SPI_PREINIT_IPU 2 A15
resource SPI_PREINIT_IPU 3 B12
resource RX_SPI_CS 1 A15

i hope somebody helps me

by the way LED_Strip was A00 and i changed to be my MOTOR 2 .. and the motors work just fine .. all my problem is on ADC_BATT

thanks

nerdCopter, 2021-05-24:
i tried this in `src/main/target/MATEKF411RX/target.h`
```
#if !defined(CRAZYBEEF4FS) || !defined(CRAZYBEEF4FR) || !defined(CRAZYBEEF4DX)
#define ADC_BATT 1 A01
#endif
```

i'm not the target guy, but worth a try.
[EmuFlight_0.3.4_MATEKF411RX_Build_e75a2c17c.hex.zip](https://github.com/emuflight/EmuFlight/files/6532726/EmuFlight_0.3.4_MATEKF411RX_Build_e75a2c17c.hex.zip)

Nalkhashab, 2021-05-24:
it doesn't work :(

nerdCopter, 2021-05-26:
confusing that BF unified show B00/B01 as well.
https://github.com/betaflight/unified-targets/blob/master/configs/default/MTKS-MATEKF411RX.config
```
resource ADC_BATT 1 B00
resource ADC_CURR 1 B01
```
unsure how A01 is being applied in BF.
i also noticed `resource GYRO_EXTI 1 A01`

thefaba, 2021-06-01:
hi,

ive checked that with my MATEKF411RX on emuflight 0.3.3 and 0.3.4:
My voltage meter, works like a charm on 1S and 2S - on both firmware versions.

So i dont think, thats a problem in general with emuflight or a specific version ...

here are my settings, which are from the default:

version
EmuFlight / MATEKF411RX (M41R) 0.3.4 Apr 16 2021 / 19:07:32 (1014a9331) MSP API: 1.50
resource ADC_BATT 1 B00
resource ADC_RSSI 1 NONE
resource ADC_CURR 1 B01
resource ADC_EXT 1 NONE

set vbat_scale = 110
set vbat_divider = 10
set vbat_multiplier = 1

closed 2022-08-08 by nerdCopter