emuflight/EmuFlight #586: [Target Request] GEPRCF711

0xPr0xy, 2021-05-26:
Hi there @nerdCopter !

I'm wondering how I can help getting support for GEPRCF711 as I'd really like to try out EmuFlight on my quad.
I see that GEPRCF711_PRO is supported but I tried it and it doesn't work., not sure what is different between the two or how I can help.

Crossing fingers here hoping  it is easy to add support because the _PRO already exists..

nerdCopter, 2021-05-26:
we only have one guy making targets and he's back-logged.
basically the target definitions need to be made.  e.g. files in `src/main/target/`

0xPr0xy, 2021-05-26:
Thanks for the reply! 
I have checked out Betaflight source but it seems they are using something called "unified target configuration". basically config files read in to then make up such a target definition. (question: I guess Emuflight is not merging in this "unified target configuration"?)  
I see some similarities between the [target definition in Emuflight](https://github.com/emuflight/EmuFlight/blob/master/src/main/target/GEPRCF411_PRO/target.h) and the combination of [config file](https://github.com/betaflight/unified-targets/blob/master/configs/default/GEPR-GEPRCF411_PRO.config) and [generic target](https://github.com/betaflight/betaflight/blob/master/src/main/target/STM32F411DISCOVERY/target.h) in Betaflight so that's a start.
I will dig in a little deeper to see if I can contribute, but I don't feel to confident 😬.
Any tips or a push in the right direction on how to proceed from here to a PR would be welcome (I hope I'm even looking in the right places).

0x

nerdCopter, 2021-05-27:
right, the 0.X base is not unified, but the 1.0 re-base will be.  however the 1.0 is very far from release.
Someone has to manually create targets until then. :(

BeauBrewski, 2021-06-29:
Is this a typo and meant to be GEPRC F411 ?

Also, what flight controller is this off of, as the current line of GEPRC F411 flight controllers I'm seeing use the MatekF411 target.

0xPr0xy, 2021-06-29:
@BeauBrewski Indeed it was a typo.

```
# board_name
board_name GEPRCF411

# version
# Betaflight / STM32F411 (S411) 4.2.9 Apr 27 2021 / 19:33:23 (e097f4ab7) MSP API: 1.43
# config: manufacturer_id: GEPR, board_name: GEPRCF411, version: 8758488d, date: 2019-12-29T10:31:32Z
# board: manufacturer_id: GEPR, board_name: GEPRCF411
```

Quad specific:

```
Name: TinyGO
Brand: GEPRC
Frame: GEP-TG16
Motor to Motor: 79mm
FC: GEP-12A-F4
MCU: STM32F411
IMU: MPU6000(SPI)
OSD: BetaFlight OSD w/ AT7456E chip
ESC: BLheli_S 12A
VTX: 5.8G 25~200mW VTX
Camera: Caddx Loris 4K 60fps
Propeller: GEMFAN 1636*4 (4 pairs)
Motors: GR1102 10000Kv
TinyGO 4K Weight: 54.8 Grams
```

nerdCopter, 2021-06-29:
* looks like you can flash `MATEKF411` and then you are good to go, @0xPr0xy .
* worst case is that you need to hol the boot-button to reflash betaflight.
* of course save your `diff`/`dump` first.

0xPr0xy, 2021-06-30:
I have a strange problem after flashing.

After flashing I did:
- paste relevant bits of the diff to get my ports and things setup
- then checked the Quad settings and Motor direction settings, and verified this from Emuflight by running each motor
- Checked the rest

Then turned on my Quad and right away my HomePod speakers disconnected from the TV (bluetooth) 😄 You'd think that be related to the VTX but I did not have this with Betaflight. very weird (I pasted the VTX table in after but that doesn't change anything). Anyway strange but I decided to give flying a go.

For some reason it started it crash flip mode so I changed that to be off by default.
Then Armed, all motors spinning, put throttle and I crash flip! Each time! crash flip enabled or not. I do see the OSD toggling when I change aux3 and I know I set aux3 to not have crash flip on all 3 states.
So I'm lost.. what would cause this? motors all work fine in EF, and all spin when arming. only throttle input is suddenly making it flip.

nerdCopter, 2021-06-30:
* EmuFlight does not have VTX tables.
* EmuFlight 0.X cannot take BF 4.X unified-target resources.
* Many BetaFlight CLI variables do not necessarily coincide with EmuFlight CLI variables.
* cli" `get vtx_` for some options. requires smartAudio configured and working.
* VTX is 5.8Ghz range.  BlueTooth is 2.4Ghz range.
* your gyro orientation may be incorrect.
* your motor resources may be incorrect
* the FC target matekF411 may be incorrect after all :man_shrugging:

BeauBrewski, 2021-07-03:
-Make sure your radio setup correct AETR vs TAER. Sounds like your channel mapping got switched. Check all the switches on the receiver tab and MODES tab to make sure they are correct.
-Depending on your radio and protocol, it could operate on 2.4ghz and cause some weird voodoo with Bluetooth electronics? 
-Worst case, reflash `MATEKF411` to default and setup quad from scratch.

closed 2021-11-13 by BeauBrewski