emuflight/EmuFlight #589: OSD Languages

JulioCesarMatias, 2021-05-28:
hello, in the future will this be implemented in Emu? I was thinking of adding Brazilian Portuguese translations. Are you using # define, in the future will a Box be implemented to select languages?

https://github.com/emuflight/EmuFlight/pull/490/files

Quick-Flash, 2021-06-07:
> hello, in the future will this be implemented in Emu? I was thinking of adding Brazilian Portuguese translations. Are you using # define, in the future will a Box (in EmuConfigurator) be implemented to select languages?
> 
> https://github.com/emuflight/EmuFlight/blob/memory-management/src/main/osd/osd_languages.h

just noticed this, my bad. Yes this is our plan, I haven't finished all the code for it yet though, it'll take a bit of work to make all the osd options into defines.  Once its done though an option to select a language in the configurator will be added.

JulioCesarMatias, 2021-06-07:
maybe creating a primary and a secondary table with the translations wouldn't be better?

 for example:

https://pastebin.com/hFXLRAMp

i tested in the avr  and stm32 platforms

> just noticed this, my bad. Yes this is our plan, I haven't finished all the code for it yet though, it'll take a bit of work to make all the osd options into defines. Once its done though an option [...]

JulioCesarMatias, 2021-06-07:
some words are also bad to translate, because in Portuguese they are bigger than in English, and it ends up not fitting in the OSD

Quick-Flash, 2021-06-07:
>  maybe creating a primary and a secondary table with the translations wouldn't be better?
> 
>  for example:
> 
> https://pastebin.com/hFXLRAMp
> 
> i tested in the avr  and stm32 platforms
> 

Would be better except that we are already getting tight with memory and this would make us totally run out. Each language added would use more memory. As for some things not translating nicely perhaps some things can stay english???

JulioCesarMatias, 2021-06-07:
> Would be better except that we are already getting tight with memory and this would make us totally run out. Each language added would use more memory. As for some things not translating nicely [...]

yes, it should be in english. I made some changes (See in pastebin). Perhaps this should only be applied in FC with uC F4,F7 and H7. For F3, leave everything in English.

Quick-Flash, 2021-06-07:
> > > maybe creating a primary and a secondary table with the translations wouldn't be better?
> > > for example:
> > > https://pastebin.com/hFXLRAMp
> > > i tested in the avr  and stm32 platforms
> > 
> > Would be better except that we are already getting tight with memory and this would make us totally run out. Each language added would use more memory. As for some things not translating nicely perhaps some things can stay english???
> 
> yes, they must be in english. I made some changes (See in pastebin). Perhaps this should only be applied in FC with uC F4,F7 and H7. For F3, leave everything in English.

Flash a stm32f411 target in a 1.0.0 branch and you will sadly see how tight memory is.

JulioCesarMatias, 2021-06-07:
> Flash a stm32f411 target in a 1.0.0 branch and you will sadly see how tight memory is.

you're right!Sorry.

Quick-Flash, 2021-06-07:
No worries. If you can think of a different better way to handle it let me know. Or if you come up with a great idea for memory management that would be great.

JulioCesarMatias, 2021-06-08:
> No worries. If you can think of a different better way to handle it let me know. Or if you come up with a great idea for memory management that would be great.

unfortunately i don't know how to optimize this, if i have a shape it is far from my knowledge. The only way out I think at the moment is to provide Emu patches with translations, and let the user choose a .hex according to their local language, or if that's not nice, it's better to leave everything in English.

closed 2021-06-14 by JulioCesarMatias