emuflight/EmuFlight #592: ZEEZF7V2

TTGG4400, 2021-06-01:
Vendor/Model

- ZEEZF7V2
- ZEEZ RC F7 3030 FLIGHT CONTROLLER V2
- https://zeezrc.com/shop/zeez-f7-flight-controller/

CLI dump or Unified Target

- https://github.com/betaflight/unified-targets/blob/master/configs/default/ZEEZ-ZEEZF7V2.config

TTGG4400, 2021-06-01:
this was entered as a target request - dunno why it doesn't say that though

BeauBrewski, 2021-08-11:
Needs tested. Check orientation, motors, OSD, all that other stuff. PROPS OFF!
https://cdn.discordapp.com/attachments/547211754845765639/874799667429396490/EmuFlight_0.3.7_ZEEZ_ZEEZF7V2_Build_uncommitted_20210810EDT.hex

TTGG4400, 2021-08-11:
Unsure of my part in this, if any - should I install this version and test it on my 5" quad that hosts this FC?

nerdCopter, 2021-08-12:
[EmuFlight_0.3.4_ZEEZ_ZEEZF7V2_Build_5fb813e43.hex.zip](https://github.com/emuflight/EmuFlight/files/6976420/EmuFlight_0.3.4_ZEEZ_ZEEZF7V2_Build_5fb813e43.hex.zip)

TTGG4400, 2021-08-13:
I installed EmuFlight_0.3.4_ZEEZ_ZEEZF7V2_Build_5fb813e43.hex.zip and all is working except haven't been able to arm since I have a GPS - and I cannot find a CLI cmd to ignore GPS.

TTGG4400, 2021-08-13:
So - I disabled the GPS - and it armed, and the motors behaved as they should when I moved it about in my hand.

nerdCopter, 2021-08-13:
@TTGG4400 , thank you!
here is the GPS unlock: https://github.com/emuflight/EmuFlight/wiki/GPS-Lock-Bypass

TTGG4400, 2021-08-13:
@nerdCopter  Thx for that workaround.  It's perfect.  Let's me wait for sats - or not.

TTGG4400, 2021-08-13:
It flies nicely too - just got in 2 min before the rains began again.  Thx for the SW.

Rubber80, 2021-08-14:
Buongiorno Ragazzi, come faccio per installarlo sulla mia zeez v2? lo scarico e lo carico in modalità locale?
è gia testato? scusate ma non sono molto bravo

BeauBrewski, 2021-08-14:
> Buongiorno Ragazzi, come faccio per installarlo sulla mia zeez v2? lo scarico e lo carico in modalità locale?
> è gia testato? scusate ma non sono molto bravo

Baricare in modalità locale. Non ancora testato.
[image]

Rubber80, 2021-08-14:
ok grazie. Quando uscirà la versione definitiva?

BeauBrewski, 2021-08-14:
Dopo aver testato da generosi volontari 😄

TTGG4400, 2021-08-14:
And today I got in a full flight using  EmuFlight_0.3.4_ZEEZ_ZEEZF7V2_Build_5fb813e43.hex.zip.  Love it.  The Baro and homing arrow and distance home were all spot on for the whole flight.  The 5" quad felt good on the default PIDs that I loaded from the 5" build em from scratch PID presets.  And 50% expo on the sticks.

Rubber80, 2021-08-15:
ragazzi, purtroppo la vtx non trasmette nessun segnale agli occhiali fpv. cosa posso fare per correggere l'errore?

nerdCopter, 2021-08-15:
does BlackBox work? do LEDs work?

Rubber80, 2021-08-15:
ho appena fatto tre lipo, che dire.... stupendo come sempre!!!Ho volato con Pid stock ed eliche rovinate,il Propwash era quasi nulla!! Complimenti! l'unica cosa che sembra non funzionare è "vtx pit mode" in modalità di volo.

TTGG4400, 2021-08-15:
> * does BlackBox work? do LEDs work?
> * i'll be ready to merge to master, if good.

Blackbox works fine.  I grabbed it using the FC in storage mode, and read it with BF BBLE.  I do not fly with LEDs - so cannot comment on that.

Rubber80, 2021-08-15:
ho configurato i led sull'arm, e cambiano colore nel momento in cui armo

nerdCopter, 2021-08-15:
closed by #669

closed 2021-08-15 by nerdCopter