emuflight/EmuFlight #607: Mapping LED_Strip resource bricks JBF7 (same issue as #334)

VicVoss, 2021-06-15:
**Describe the bug**
Just like #334 mapping the LED_STRIP resource (which is missing) on the JBF7 target causes it to brick and must be reflashed.

**To Reproduce**

1. Flash Emuflight JBF7 0.3.4
2. Map via: resource led_strip 1 A01
3. Enable LED_STRIP in configurator
4. FC is bricked upon reboot

**Expected behavior**
Since the resource pins are identical in betaflight, the LED strip should work.

**Flight controller configuration**
The aforementioned change is the only diff

nerdCopter, 2021-06-15:
i wonder if @joshuabardwell could help directly [ponders wildly].

nerdCopter, 2021-06-18:
https://github.com/emuflight/EmuFlight/pull/611

nerdCopter, 2021-06-18:
@VicVoss , please test and report back :) [EmuFlight_0.3.5_IFRC_JBF7_Build_bc6134d11.hex.zip](https://github.com/emuflight/EmuFlight/files/6677183/EmuFlight_0.3.5_IFRC_JBF7_Build_bc6134d11.hex.zip)

nerdCopter, 2021-08-21:
@VicVoss , any feedback about LED working now?
here is 0.4.0 as well. need feedback or cannot merge to master code.
[EmuFlight_0.4.0_IFRC_JBF7_Build_7c30b8871.hex.zip](https://github.com/emuflight/EmuFlight/files/7026206/EmuFlight_0.4.0_IFRC_JBF7_Build_7c30b8871.hex.zip)

closed 2021-09-14 by nerdCopter