emuflight/EmuFlight #612: CRAZYBEEF4SX1280

BeauBrewski, 2021-06-19:
**Vendor/Model**
CRAZYBEEF4SX1280

Happymodel ELRS F4 2G4 AIO 5in1 Flight controller built-in SPI 2.4GHz ELRS RX

http://www.happymodel.cn/index.php/2021/05/19/happymodel-elrs-f4-2g4-aio-5in1-flight-controller-built-in-spi-2-4ghz-elrs-rx/

[ELRS-F4-2G4_Betaflight_4.3.0_STM32F411_1.0.0-RC5-and-Diff-Document.zip](https://github.com/emuflight/EmuFlight/files/6681262/ELRS-F4-2G4_Betaflight_4.3.0_STM32F411_1.0.0-RC5-and-Diff-Document.zip)
http://www.happymodel.cn/wp-content/uploads/2021/05/ELRS-F4-2G4_Betaflight_4.3.0_STM32F411_1.0.0-RC5-and-Diff-Document.zip

xwings, 2021-06-21:
There is a PR in betaflight repo. 

https://github.com/betaflight/betaflight/pull/10788/files

JamesLarson, 2021-10-17:
Would the betaflight PR cover the new betaFPV ELRS spi flight controllers also?

ksyz, 2021-12-14:
> Would the betaflight PR cover the new betaFPV ELRS spi flight controllers also?

Yes, elrs-wise. But betafpv spi elrs uses different target specification. Linked in that PR.

closed 2024-04-19 by nerdCopter