emuflight/EmuFlight #625: [Target Request] HOBBYWING_XROTORF7CONV

BeauBrewski, 2021-06-28:
**Vendor/Model**
HOBBYWING_XROTORF7CONV

Hobbywing XRotor F7 Flight Controller 
https://www.hobbywingdirect.com/collections/xrotor-flight-controller-f7

**CLI dump or Unified Target**
https://github.com/betaflight/unified-targets/blob/master/configs/default/HOWI-HOBBYWING_XROTORF7CONV.config

BeauBrewski, 2021-07-20:
Saw a post on the fb page where FlywooF7dual works with this flight controller.

BeauBrewski, 2021-07-25:
Needs tested. 
https://cdn.discordapp.com/attachments/691022009865338950/868940278172635136/EmuFlight_0.3.5_HOWI_HOBBYWING_XROTORF7CONV_Build_uncommitted_20210725EDT.hex

nerdCopter, 2021-08-24:
`HOWI_HOBBYWING_XROTORF7CONV` was sent to EmuConfigurator for 0.4.0

closed 2021-08-24 by nerdCopter