emuflight/EmuFlight #628: [Target Request] GEPRCF405

BeauBrewski, 2021-06-29:
**Vendor/Model**
GEPRCF405

GEP-F405 HD Flight Controller
https://geprc.com/product/gep-f405-hd-flight-controller/

**CLI dump or Unified Target**
https://github.com/betaflight/unified-targets/blob/master/configs/default/GEPR-GEPRCF405.config

BeauBrewski, 2021-11-17:
Needs tested!
Check gyro orientation, motor arrangement, flash, osd, vtx control, everything else if possible please update here
PROPS OFF FOR SAFETY

[EmuFlight_0.4.0_GEPR_GEPRCF405_Build_bd7a38d66.zip](https://github.com/emuflight/EmuFlight/files/7557333/EmuFlight_0.4.0_GEPR_GEPRCF405_Build_bd7a38d66.zip)

nerdCopter, 2022-07-27:
tested by @captainvietnam, all good. except LEDs not tested.
```
2022-06-26 [5:45 PM CDT]  CaptainVietnam6: @Beau Brewski @nerdCopter just tested the target for GepRCF405
Flash, gyro, accel, motors, ADCs, OSD, SA control, Beeper, barometer, blackbox, GPS works
unable to test LEDs 
```

PR #836 (2022-07-29): TARGET GEPRCF405
Tested per https://github.com/emuflight/EmuFlight/issues/628
Closes #628
closed 2022-07-29 by nerdCopter