emuflight/EmuFlight #636: [target request] JHEMCU GHF420AIO

tzshin, 2021-07-05:
**Vendor/Model**
JHEF405/JHEMCU GHF420AIO 35A
http://www.jhemcu.com/e_productshow/?39-JHEMCU-JHEMCU-GHF420AIO-35A-39.html

**CLI dump or Unified Target**
https://github.com/betaflight/unified-targets/blob/master/configs/default/JHEF-JHEF405.config

BeauBrewski, 2021-07-05:
According to the vender link you sent, the Firmware is `BETAFLIGHTF4`
[image]

Try flashing the EmuFlight `BETAFLIGHTF4` target and see if that works for you. (Props off, check orientation, etc)

Let us know if that works.

tzshin, 2021-07-12:
Haven't been able to build and test the whole quad due to slow ships. But with slight comparison between BETAFLIGHTF4 config and cli dump, it should be right one.

The default firmware on the fc seems to be a custom (or modified) one, tagged with a confusing unique firmware target.

soukron, 2021-09-11:
FWIW I flashed betaflightf4 target and even when apparently everything works (motors, osd, gyro, sbus, etc) I can't arm nor see the voltage of the lipo in the OSD. I must be missing something or better yet the target isn't the right one.

BeauBrewski, 2021-09-13:
> FWIW I flashed betaflightf4 target and even when apparently everything works (motors, osd, gyro, sbus, etc) I can't arm nor see the voltage of the lipo in the OSD. I must be missing something or better yet the target isn't the right one.

@soukron Got a copy of your BF and Emu dumps for your build?

closed 2021-11-13 by BeauBrewski