emuflight/EmuFlight #639: [Target Request] JHEMCU GHF405 PRO

R2D2FISH, 2021-07-08:
This is a bit of a weird one... according to most vendors, the correct target is "JHE405", however, no such target exists.  There does exist a "JHEF405" target, but it does not match the config which comes preinstalled on the board.  According to the preinstalled config, its target is supposed to be called "JHE405PRO".  This CLI dump works on my own hardware (the JHEF405 unified target does not!)

**Vendor/Model**
JHE405PRO/JHEMCU GHF405 PRO (25/45A)
https://www.team-blacksheep.com/products/product:5259 (the board is not listed on the JHEMCU website, for some reason)

**CLI dump or Unified Target**
```
board_name JHE405PRO
manufacturer_id JHEM

# resources
resource BEEPER 1 C13
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 B05
resource MOTOR 6 B07
resource MOTOR 7 C09
resource MOTOR 8 C08
resource PPM 1 B08
resource LED_STRIP 1 A09
resource SERIAL_TX 1 B06
resource SERIAL_TX 2 D05
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 D06
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 6 C07
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09
resource LED 1 C14
resource SPI_SCK 1 A05
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 B08
resource ADC_BATT 1 C03
resource ADC_RSSI 1 C00
resource ADC_CURR 1 C02
resource FLASH_CS 1 B03
resource OSD_CS 1 B14
resource GYRO_EXTI 1 B13
resource GYRO_CS 1 B12
resource USB_DETECT 1 A08

# timer
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer A03 AF1
# pin A03: TIM2 CH4 (AF1)
timer A02 AF1
# pin A02: TIM2 CH3 (AF1)
timer B05 AF2
# pin B05: TIM3 CH2 (AF2)
timer B07 AF2
# pin B07: TIM4 CH2 (AF2)
timer C09 AF3
# pin C09: TIM8 CH4 (AF3)
timer C08 AF3
# pin C08: TIM8 CH3 (AF3)
timer A09 AF1
# pin A09: TIM1 CH2 (AF1)

# dma
dma ADC 1 0
# ADC 1: DMA2 Stream 0 Channel 0
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin A03 1
# pin A03: DMA1 Stream 6 Channel 3
dma pin A02 0
# pin A02: DMA1 Stream 1 Channel 3
dma pin B05 0
# pin B05: DMA1 Stream 5 Channel 5
dma pin B07 0
# pin B07: DMA1 Stream 3 Channel 2
dma pin C09 0
# pin C09: DMA2 Stream 7 Channel 7
dma pin C08 0
# pin C08: DMA2 Stream 2 Channel 0
dma pin A09 0
# pin A09: DMA2 Stream 6 Channel 0

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature MOTOR_STOP
feature RSSI_ADC
feature LED_STRIP
feature OSD

# led
led 0 4,6::CB:8
led 1 5,6::CB:8
led 2 6,6::CB:8
led 3 7,6::CB:8

# master
set mag_bustype = I2C
set mag_i2c_device = 1
set mag_hardware = NONE
set baro_bustype = I2C
set baro_i2c_device = 1
set serialrx_provider = SBUS
set blackbox_device = SPIFLASH
set dshot_burst = ON
set motor_pwm_protocol = DSHOT600
set current_meter = ADC
set battery_meter = ADC
set ibata_scale = 170
set beeper_inversion = ON
set beeper_od = OFF
set osd_vbat_pos = 14571
set osd_ah_sbar_pos = 14542
set osd_ah_pos = 14414
set osd_current_pos = 14634
set system_hse_mhz = 8
set max7456_spi_bus = 3
set dashboard_i2c_bus = 1
set pinio_box = 40,41,255,255
set flash_spi_bus = 3
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_sensor_align = CW180
set gyro_1_align_yaw = 1800
set gyro_2_spibus = 1
save
```

BeauBrewski, 2021-11-12:
@R2D2FISH I used your diff and compared it with the BF config to build this. I added the 'F' to the target name because its an F405 board and I refuse to perpetuate a typo of that magnitude. 😆 

Needs tested!
Check gyro orientation, motor arrangement, flash, osd, vtx control, everything else if possible please update here
PROPS OFF FOR SAFETY

[EmuFlight_0.4.0_JHEF405PRO_Build_b4edd88e0.zip](https://github.com/emuflight/EmuFlight/files/7524707/EmuFlight_0.4.0_JHEF405PRO_Build_b4edd88e0.zip)

Peryshev, 2021-11-12:
Thanks , will test it today

BeauBrewski, 2022-03-11:
@Peryshev @R2D2FISH 
Any feedback? Had a gyro issue that was fixed early January, test most updated zip file that is posted above.
Feedback is much appreciated for as many features as possible or available:
✔️/❌ Flash
✔️/❌ Gyro
✔️/❌ Accelerometer
✔️/❌ Motors
✔️/❌ Onboard ADC
✔️/❌ OSD (analog)
✔️/❌ Smart Audio controls (OSD and CLI)
✔️/❌ Alarm switch
✔️/❌ Barometer
✔️/❌ Blackbox
✔️/❌ LED
✔️/❌ GPS
Or any other features or issues you notice

BeauBrewski, 2022-05-08:
Tested:
✔️ Flash
✔️ Gyro
✔️ Accelerometer
✔️ Motors
✔️ Onboard ADC
✔️ OSD (analog)
✔️ Smart Audio controls (OSD and CLI)
? Alarm switch
? Barometer
✔️ Blackbox
? LED
? GPS

BeauBrewski, 2022-05-09:
Baro works in BF, not in Emu. Double checking code and trying to find out what baro is on this board

aureloctopuzzle, 2025-01-13:
hello, please can you tell me where is the most updated zip file that is posted above? i dont see it..
i dont know why.
thank you! i have two of this board, and no target for BF in 4.5 (or 4.3) and is with 4.2 now..
AureL.

nerdCopter, 2025-01-13:
EmuFlight:
[image]

BetaFlight:
[image]

aureloctopuzzle, 2025-01-15:
I think this is all due to a fault typo on the keyboard.

nerdCopter, 2025-01-15:
Emuflight did not always adhere to Betaflight naming convention.
Emuflight does not support DPS310.
`F`mean nothing. 405 alludes to `STM32F405`. 405 or F405 is the same chip.

aureloctopuzzle, 2025-01-16:
> * Emuflight ne prend pas en charge DPS310.
> * Emuflight n'a pas toujours respecté la convention de dénomination de Betaflight. `F`Cela ne veut rien dire. 405 fait allusion à `STM32F405`. 405 ou F405 est la même puce.
> 
> De plus, je suis désolé, je vois cela maintenant dans EmuFlight : il est donc plus probable que ce soit celui que vous voulez. N'essayez pas [image]`JHEF405PRO``JHEMCU405`
> 
> [modifier : confirmé `JHEF405PRO`qu'il dispose des mêmes ressources que le fichier de définition de Betaflight. désolé pour toute confusion]

aureloctopuzzle, 2025-01-16:
hello, it would be for betaflight 4.5, because I don't know emuflight.. sorry. and I prefer to use software that I know well. anyway, so I'll try, as soon as I get home (bf 4.5) I hope it will work.. thanks for your messages, I'll tell you about the results.
cordially,
AureL.

nerdCopter, 2025-01-16:
> hello, it would be for betaflight 4.5, because I don't know emuflight.. sorry. and I prefer to use software that I know well. anyway, so I'll try, as soon as I get home (bf 4.5) I hope it will work.. thanks for your messages, I'll tell you about the results. cordially, AureL.

okay, well this is `Emuflight` github -- will not support betaflight issues here.

aureloctopuzzle, 2025-01-18:
ok if is ok is possible i test emuflight. is for a 2.5 inch cinewhoop (foxeer unbreakable). do you think is ok for it?
thank a lot!
AureL