emuflight/EmuFlight #640: Emuflight killed 2 Mamba F411 aio boards

ChasedMonkey, 2021-07-10:
Last week I installed JESC and Emuflight on my tinywhoop board. After getting everything setup in emuflight and configuring the motor directions in jesc configurator. I went to test hover, the quad tried to runaway and shutdown. After that motor 2 wouldn't spin any more. I connected to emuflight and motor 2 wouldn't spin but the rest would. I went over to JESC configurator and only ESC 1,3 and 4 show up, there is no way to get esc 2 to show up in JESC configurator. I thought I fried the esc with JESC so I ordered a new Mamba F411 aio.

The new mamba F411 AIO got here today. I set it up exactly like I tried to last time but used Bluejay instead. Same thing happened after that. I installed Bluejay on the esc, setup emuflight, configured my motor direction in emuflight and in bluejay. I went to test hover and the quad tried to flip out. I tried a second test hover and motor 2 didn't spin. Now I can't get ESC 2 to display in Bluejay, and motor 2 is non responsive in emuflight motors page. I did try to test hover with goggles this time and for some reason my battery individual cell voltage displayed in the osd as o.ov. 

Please tell me there is a fix and that I'm not just out 2 flight controllers.

**Flight controller configuration**
# version
# EmuFlight / MATEKF411 (MK41) 0.3.4 Apr 16 2021 / 19:06:49 (1014a9331) MSP API: 1.50

board_name 
manufacturer_id 

# name
name InGageFPV

# resources

# mixer

# servo

# servo mix

# feature
feature -AIRMODE

# beeper

# beacon

# map

# serial
serial 1 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 900 1125 0 0
aux 1 2 2 1825 2100 0 0
aux 2 28 2 900 1175 0 0
aux 3 28 2 1825 2100 0 0
aux 4 35 3 900 1150 0 0

# adjrange

# rxrange
rxrange 1 1000 2004
rxrange 2 988 2011
rxrange 3 1000 2011

# vtx

# rxfail

# master
set gyro_sync_denom = 2
set baro_hardware = NONE
set serialrx_provider = CRSF
set motor_pwm_protocol = DSHOT300
set vbat_max_cell_voltage = 44
set vbat_min_cell_voltage = 32
set yaw_motors_reversed = ON
set osd_vbat_pos = 449
set osd_crsf_rssi_pos = 2101
set osd_flymode_pos = 2081
set osd_current_pos = 2518
set osd_craft_name_pos = 2505
set osd_warnings_pos = 2153
set osd_avg_cell_voltage_pos = 2497

# profile
profile 0

# rateprofile
rateprofile 0

**Setup / Versions**
 - Flight controller Diatone Mamba F411 AIO 26.5mm
 - Other components Diatone Hey Tina Whoop stock parts
 - Factory wiring

nerdCopter, 2021-07-10:
do you know 100% `MATEKF411` is the target? getfpv reports `MAMBAF411` is target name for BF. We probably do not have mambaF411, but not 100% certain.

ChasedMonkey, 2021-07-10:
> do you know 100% `MATEKF411` is the target? getfpv reports `MAMBAF411` is target name for BF. We probably do not have mambaF411, but not 100% certain.
> 
> Although Matek F411 is a target for many vendors out there, i'm not sure about this "new" hardware.

I'm an idiot. Thanks

closed 2021-07-26 by nerdCopter

gretel, 2021-08-24:
`Resolved` is lacking any semantics. please use proper language not asking to ask.