emuflight/EmuFlight #641: Zeez F7 Flight Controller - No on-board/flash storage option

bigjoncoop, 2021-07-12:
Zeez F7 Flight Controller (Non HD Version) 
Firmware Version: EmuFlight_0.3.4_ZEEZF7.hex
Configurator Version: 0.3.4
Windows PC

In the Black Box tab there is no option for onboard flash storage. The only option is serial.

This flight controller has 128MB of on-board flash Storage.

If I flash Betaflight back on to it using the same Zeez F7 target, the on-board flash storage option is available in the Black Box tab. It also functions properly. (Black Box data is stored there and the onboard flash memory becomes available has a mass storage device) 

When I get home I will do a /diff and post the contents here. But everything should be stock since I've only made the normal changes in the configurator and haven't done anything in the CLI

nerdCopter, 2021-07-12:
This is know. drivers for 0.3.X base are not up to date and it's been attempted multiple time without success.
Efforts will not be made for 0.X code-base as eventually the fix will come from the rebase from betaflight 4.x

bigjoncoop, 2021-07-12:
Is there anything I can do in order to get the onboard flesh to work? Like rolling back to 0.3.3?

And this issue is board specific? Cuz I can swear I have other flight controllers with on-board flash that work

nerdCopter, 2021-07-12:
this issue is flight-controller-specific. Zeez and some other FC that i've forgotten which. reverting will not work, same problem exists on older versions.

BeauBrewski, 2021-07-13:
Awww, same issue I'm having with #633, but different fc.

bigjoncoop, 2021-07-13:
@nerdCopter 
Ok, Thanks for the info. I guess ill have to install a serial logger (open logger)

normally not having a black box isn't a issue but this is a new 10in build and definitely need some black box data.

nerdCopter, 2021-07-13:
i almost replied to get LPF's from a BF flash, but o recanted because i figured with rpm filtering, etc there would be no noise profile to review. (i.e. would have to turn off a bunch f features to get true-noise).  i think soldered logger is the way to go here.  fwiw, we did repeatedly try to update drivers, but it was a mess.   final solution is wait for Emu 1.0.0 :(