emuflight/EmuFlight #650: I wanna gps_rescue_allow_arming_without_fix

kaji331, 2021-07-22:
Please add this parameter in CLI. I can not find it in 0.3.4 version (MATEKF405). I think if some one needs this in CLI mode, he must know what he does.

nerdCopter, 2021-07-22:
@kaji331 , 
https://github.com/emuflight/EmuFlight/wiki/GPS-Lock-Bypass
does this help?

kaji331, 2021-07-22:
> @kaji331 ,
> https://github.com/emuflight/EmuFlight/wiki/GPS-Lock-Bypass
> does this help?

Thank you very much. I read it, however it confused me. If I set gps_rescue_min_sats = 0, I know I can arm the quad. But I don't know whether I can go home when the sats > 6 or not. If I can, where is my home when I am flying? Or set gps_rescue_min_sats = 0 is equal to set gps_rescue_allow_arming_without_fix = ON in betaflight? When I arm my quad with sats < 6, if the gps rescue mode is off in this time flying even I get sats > 6? I must disarm and arm again with sats > 6, then get gps rescue mode ON? Sorry to my English!

nerdCopter, 2021-07-22:
i never tried gps, but from my understanding:
 * set gps_rescue_min_sats = 0 is equal to set gps_rescue_allow_arming_without_fix = ON
 * cannot return home, unless sats >=6 **_before arming_**

maybe join discord and discuss with @davesbiker for additional info

kaji331, 2021-07-22:
Thank you very mush. I think maybe I need some tests. Now I understand the mechanism is: I set gps_rescue_min_sats = 0, I will arm my quad without sats and I can turn on gps rescue mode but I would not get my quad back home correctly even other place since sats > 6 in the fly. I think this mechanism may be more risk than gps_rescue_allow_arming_without_fix = ON, because the movement of quad would not be predicted. Or more details wiki needed.

nerdCopter, 2021-07-22:
agree.  Emu has not modified much of old GPS code from butterflight/betaflight 3.5.x .

closed 2021-09-14 by nerdCopter