emuflight/EmuFlight #652: GPS Rescue sometimes turns off the motors & Gyro issue

aLDime, 2021-07-25:
**Describe the bug**
GPS Rescue sometimes turns off the motors & Gyro issue

**To Reproduce**
If RXLOSS--->RTH Rescue 

**Expected behavior**
Guaranteed Rescue with RXLOSS and correct gyro operation 

**Flight controller configuration**
`# diff all

# version
# EmuFlight / FLYWOOF411 (FW41) 0.3.4 Apr 16 2021 / 19:01:24 (1014a9331) MSP API: 1.50

board_name 
manufacturer_id 
mcu_id 0029005a3130511730343830
signature 

# reset configuration to default settings
defaults nosave

# name
name Mark4_7inch

# resources

# mixer

# servo

# servo mix

# feature
feature -SOFTSERIAL
feature -TELEMETRY
feature -LED_STRIP
feature -AIRMODE
feature MOTOR_STOP
feature GPS

# beeper
beeper -RX_LOST_LANDING
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED
beeper -RC_SMOOTHING_INIT_FAIL

# beacon

# map

# serial
serial 0 2 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 3 1700 2100 0 0
aux 1 1 4 900 1400 0 0
aux 2 46 1 1700 2100 0 0
aux 3 13 6 1700 2100 0 0
aux 4 19 5 1700 2100 0 0
aux 5 28 4 1700 2100 0 0

# adjrange
adjrange 0 0 3 900 1200 0 3 0 0

# rxrange

# vtx

# rxfail
rxfail 4 s 1000
rxfail 6 s 1000
rxfail 7 s 2000
rxfail 8 s 1000
rxfail 9 s 1000
rxfail 10 s 1000
rxfail 11 s 1000
rxfail 12 s 1000
rxfail 13 s 1000

# master
set gyro_lowpass2_hz_roll = 300
set gyro_lowpass2_hz_pitch = 300
set gyro_lowpass2_hz_yaw = 300
set gyro_notch1_hz = 400
set gyro_notch2_hz = 200
set acc_calibration = -26,54,122
set mag_hardware = NONE
set max_check = 1950
set rssi_channel = 5
set serialrx_provider = IBUS
set blackbox_p_ratio = 16
set min_throttle = 1030
set dshot_idle_value = 500
set failsafe_switch_mode = STAGE2
set failsafe_procedure = GPS-RESCUE
set vbat_max_cell_voltage = 44
set vbat_min_cell_voltage = 29
set vbat_warning_cell_voltage = 31
set vbat_scale = 111
set ibata_scale = 175
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_rescue_angle = 40
set gps_rescue_initial_alt = 70
set gps_rescue_descent_dist = 30
set gps_rescue_ground_speed = 1500
set gps_rescue_throttle_max = 1700
set gps_rescue_throttle_hover = 1450
set gps_rescue_sanity_checks = RESCUE_SANITY_OFF
set gps_rescue_min_sats = 6
set osd_warn_arming_disable = ON
set osd_warn_batt_not_full = ON
set osd_warn_batt_warning = ON
set osd_warn_batt_critical = ON
set osd_warn_visual_beeper = ON
set osd_warn_crash_flip = OFF
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = ON
set osd_warn_rc_smoothing = OFF
set osd_warn_dji = OFF
set osd_rssi_alarm = 25
set osd_cap_alarm = 3500
set osd_alt_alarm = 150
set osd_tim1 = 5120
set osd_tim2 = 3841
set osd_vbat_pos = 2295
set osd_rssi_pos = 2168
set osd_tim_1_pos = 2091
set osd_tim_2_pos = 2123
set osd_flymode_pos = 2381
set osd_throttle_pos = 2370
set osd_crosshairs_pos = 2253
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_current_pos = 2358
set osd_mah_drawn_pos = 2392
set osd_craft_name_pos = 2409
set osd_gps_speed_pos = 2338
set osd_gps_lon_pos = 2448
set osd_gps_lat_pos = 2434
set osd_gps_sats_pos = 2146
set osd_home_dir_pos = 2212
set osd_home_dist_pos = 2178
set osd_compass_bar_pos = 2154
set osd_altitude_pos = 2114
set osd_warnings_pos = 2345
set osd_avg_cell_voltage_pos = 2327
set osd_disarmed_pos = 2315
set osd_nvario_pos = 225
set osd_core_temp_pos = 2136
set osd_stat_tim_1 = ON
set osd_stat_max_dist = ON
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set osd_stat_bb_no = OFF
set vcd_video_system = NTSC

# profile
profile 0

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_expo = 50
set pitch_expo = 50
set yaw_expo = 50

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# `

**Setup / Versions**
 -  NOXE F4 V1 with baro & BB
 - ESC Racerstar RS20Ax4in1 Blheli_S; Matek FCHUB-6S; FlySky iA6B RX; GPS BEITIAN BN-220

**Additional context**
For 4 rather long flights (4xBat), ~10 times the RTH Rescue worked as expected! But at some point, just turned off motors at start of Rescue:
[image]

[EMUF_BLACKBOX_LOG_Mark4_7inch_20210723_182344.zip](https://github.com/emuflight/EmuFlight/files/6874158/EMUF_BLACKBOX_LOG_Mark4_7inch_20210723_182344.zip)

[Short video this crash](https://youtu.be/BqWbRA7T9sE)

Strange behavior gyroscope, it constantly "floats" in BF & iNav - the gyroscope on this copter works perfectly, if need I have video with it's!

[Full flight and in end crash, look as work gyro](https://youtu.be/14-ZTQZEl6E)

I really liked EmuFlight, I would like to solve this issue and it will be just great! Grateful

aLDime, 2021-07-25:
Looked at this target in BF, gyro = CW180 by default
Need to set align_gyro = CW180 right?
Issue with motor disconnection at RTH can be associated with gyro?

nerdCopter, 2021-07-26:
if gyro alignment is incorrect, then you would not be able to fly.
sorry i do not know things about GPS.  maybe have to join discord and ask if not already tried.

aLDime, 2021-07-27:
Well, I spent a few more flights, everything is clear to me, you have a BUGs with a Gyroscope, the horizon is "floating"(CW0 and CW180 - the same ), and you also have a BUG in the RTH as in the BF (or maybe even as in the CF), since the same result - may be several times work normally , but once again turns off the motors for no apparent reason, even if the flight conditions are ideal! I am deleting your firmware as isn't possible to fly LONG RANGE on it!

aLDime, 2021-07-27:
Last flights, maybe you might find it useful:
[image]
[EMUF_BLACKBOX_LOG_Mark4_7inch_20210726_190103.zip](https://github.com/emuflight/EmuFlight/files/6882532/EMUF_BLACKBOX_LOG_Mark4_7inch_20210726_190103.zip)
[EMUF_cli_Mark4_7inch_20210726_105620.txt](https://github.com/emuflight/EmuFlight/files/6882533/EMUF_cli_Mark4_7inch_20210726_105620.txt)

Quick-Flash, 2021-08-12:
> Well, I spent a few more flights, everything is clear to me, you have a BUGs with a Gyroscope, the horizon is "floating"(CW0 and CW180 - the same ), and you also have a BUG in the RTH as in the BF (or maybe even as in the CF), since the same result - may be several times work normally , but once again turns off the motors for no apparent reason, even if the flight conditions are ideal! I'm deleting your firmware as isn't possible to fly LONG RANGE on it!

I've looked at the logs and your claim of "but once again turns off the motors for no apparent reason" is simply not what the logs are telling us. What it looks like to me is a desync from a quick yaw move caused by gps RTH. That desync led to the FC shutting off some of, or maybe even all of the motors. Thats really what this issue is looking like. Your using a 20A esc on a 7inch drone. For slower cruising flight i'm sure its fine, but that abrupt yaw move is causing issues for your motor/esc combo leading to some real problems that cause crashes. Looking at the logs GPS is working as intended, the motors just aren't doing what the FC wants them to do, but they are sure trying.

closed 2022-08-08 by nerdCopter