emuflight/EmuFlight #655: [Bug Report] Enabling GPS or GPS+Telemetry in Configurator sandbags flight controller requiring reflash of firmware.

BeauBrewski, 2021-07-26:
**Describe the bug**

1. Fresh configure of MatekF405SE works until GPS is enabled in configurator and saved. Then cannot reconnect to fc unless booted in DFU mode and reflashed. Confirmed works in BF. Tried on 0.3.3 and 0.3.4. 
2. Fresh configure of KakuteF7 HD works until GPS _and_ Telemetry are both enabled in configurator and saved. One or the other on is fine. Then cannot reconnect to fc unless booted in DFU mode and reflashed. 

**To Reproduce**
1. On MatekF405SE, enable GPS in configurator, save/exit, unplug, and try to reconnect. 
2. On KakuteF7 HD, enable GPS _and_ Telemetry in configurator, save/exit, unplug, and try to reconnect. 

**Expected behavior**
Enabling of the GPS or GPS+Telemetry to save without sand bagging the flight controller, and being able to use features to fly long range with GPS and/or Telemetry. 

**Flight controller configuration**
_Need diff's from pilots reporting issue._

**Setup / Versions**
1.   FC: MatekF405SE
     GPS: Flywoo GM8 Mini V2
     Soldered to UART5
2.  FC: KakuteF7 HD

**Additional context**
1. Works in BF with no hardware changes.

nerdCopter, 2021-07-26:
so many targets to confuse.
is kakute the proper target for certain? 
we have
```
KAKUTEF7
KAKUTEF7HDV
KAKUTEF7V15  (V2 in BF)
KAKUTEF7MINIV1
KAKUTEF7MINIV2
KAKUTEF7MINIV3
```

BeauBrewski, 2021-07-26:
Attempting to verify. _Should_ be the `KAKUTEF7HDV`.

nonoreb, 2022-01-23:
Hello, 
Did you find any solution to this ? I have the same problem with KAKUTEF7V15, I have to reflash (0.4.0) because I cannot acces fc anymore after enabling GPS.

nerdCopter, 2022-01-24:
> Hello, Did you find any solution to this ? I have the same problem with KAKUTEF7V15, I have to reflash (0.4.0) because I can't acces fc anymore after enabling GPS.

try `KAKUTEF7HDV` & let us know.
  if still fails, we need to investigate target resources and we request cli dumps of BF vs Emu.

Kaiowarez, 2022-01-25:
Ahhh I was going to add GPS drivers for those targets I'll fix it
later today

nerdCopter, 2022-08-08:
Needs testing: https://github.com/emuflight/EmuFlight/pull/850 (`MATEKF405SE`)
[EmuFlight_0.4.1_MATEKF405SE_Build_ad3b78da3.hex.zip](https://github.com/emuflight/EmuFlight/files/9284598/EmuFlight_0.4.1_MATEKF405SE_Build_ad3b78da3.hex.zip)

loutwice, 2022-10-18:
> @nerdCopter @Kaiowarez  

hi buddieeees, @Kaiowarez did you get time to add gps driver for the kakute?

nerdCopter, 2022-10-20:
@loutwice , which target do you need. i suppose i can look.  I have the MATEKF405SE .zip above, that was never tested by anyone.

PR #850 (2024-03-28): fix MATEKF405SE MAG_I2C_INSTANCE
* fix MATEKF405SE by defining BARO and MAG correctly
* need backports to 0.3.x as well.
* NEEDS TESTERS, no one stepping up -- please test this zip and comment below or in #655 : 

* [EmuFlight_0.4.1_MATEKF405SE_Build_6cc68df4e.hex.zip](https://github.com/emuflight/EmuFlight/files/11398293/EmuFlight_0.4.1_MATEKF405SE_Build_6cc68df4e.hex.zip)

closes #655
closed 2024-03-28 by nerdCopter