emuflight/EmuFlight #662: ApeX F722 FC

awe755, 2021-08-03:
**Vendor/Model**
* Please provide the target name. e.g. `MAMBAF722`
* Please provide vendor and model of the FC. e.g. `Diatone MAMBA F722 MINI STACK Flight Controller`
* Please provide a link to the vendor specifications.
 e.g https://www.diatoneusa.com/store/p654/New_Diatone_MAMBA_F722_MINI_STACK_Flight_Controller_and_MAMBA_306_ESC.html

**CLI dump or Unified Target**
* Please provide a CLI `dump` of **DEFAULTS** or link to unified target.
  Find the unified targets here: https://github.com/betaflight/unified-targets/tree/master/configs/default

nerdCopter, 2021-08-03:
What is the problem exactly?
i am flying `APEXF7`, but only on analog.

awe755, 2021-08-03:
Problem module caddx vista can't on (no indicator led is on) but in
betaflight is no problem.

awe755, 2021-08-04:
Frame Rajawali Bouraq
FC Apex F722
ESC T-Motor Velox 45A
Motor T-Motor F40 2600KV
TX TBS Crossfire nano RX
Digital CADDX Vista DJI Camera
when flash whit firmware betaflight this quad is oke. all parts no one problem and normal fly.
but i flash whit firmware emuflight this quad have a problem. this promblem in Digital CADDX Vista DJI Camera can not starup an no voltage.
i just flah and no edit wiring cable my quad, please tell me what wrong in may quad or in emuflight.
and help me to fixs this problem. thanks

nerdCopter, 2021-08-04:
* You may get better help from Zohar, the owner of Apex FPV.  He can be contacted on Facebook (https://www.facebook.com/ApexFPV/) or Discord (ApexFPV#0733).
* I can only make assumptions, as i do not have DJI.  Maybe the Ports-Tab/UART is not setup?
* I have some documentation here, but this mostly only saved from external resources: https://gist.github.com/nerdCopter/ebdc7840a243aa60d6ff9680860ea66b

awe755, 2021-08-05:
I haven't found the right solution for FC Apex with DJI Vista in Emuflight firmware.
even though I really want to fly with emuflight.
what can make me finally go back to Betaflight which is definitely not a problem with FC Apex with DJI Vista.
Betaflight everything is normal and flying happy.
what's with the emufliht firmware on FC Apex with DJI Vista.
the port tab setup is all the same as the settings in Betaflight, the cable lines are also not changed but DJI Vista still doesn't work and the current voltage is gone.
please let me know if anyone can help, I'm still looking forward to flying with Emuflight.