emuflight/EmuFlight #672: [Bug Report] RUSHBLADEF7 not detecting BMP280 Barometer.

BeauBrewski, 2021-08-13:
**Describe the bug**
Flight controller supports BMP280 in Betaflight, but will not turn on in Emuflight. See bug report #415 

**To Reproduce**
Attempt to turn on BMP280 in Emuflight.

**Setup / Versions**
 - RUSHFPV RUSH BLADE F722 FLIGHT CONTROLLER FOR DJI DIGITAL FPV SYSTEM
 - BMP280 barometer
 -Affects `RUSHBLADEF7` and possibly `RUSHBLADEF7_HD` test fw.

**Additional context**
See #415

JulioCesarMatias, 2021-08-14:
I made a correction, can you test it again?

JulioCesarMatias, 2021-08-14:
https://github.com/emuflight/EmuFlight/pull/671

BeauBrewski, 2021-08-14:
[BMP280 Test_EmuFlight_0.3.7_RUSHBLADEF7_Build_2013_3da929cc6.zip](https://github.com/emuflight/EmuFlight/files/6986693/BMP280.Test_EmuFlight_0.3.7_RUSHBLADEF7_Build_2013_3da929cc6.zip)

nerdCopter, 2021-08-14:
@JulioCesarMatias , this issue ticket for `RUSHBLADEF7`, i think you meant to post in https://github.com/emuflight/EmuFlight/issues/415

JulioCesarMatias, 2021-08-15:
> @JulioCesarMatias , this issue ticket for `RUSHBLADEF7`, i think you meant to post in #415

sorry

BeauBrewski, 2021-08-15:
@JulioCesarMatias any chance of getting a `RUSHBLADEF7` fix for BMP280 too? 🙏

JulioCesarMatias, 2021-08-15:
> RUSHBLADEF7

@nerdCopter  Can I add BMP280 to HUSHBLADE 7 via PR #415? Or, should I open a new PR?

JulioCesarMatias, 2021-08-16:
> @JulioCesarMatias alguma chance de conseguir uma `RUSHBLADEF7`correção para BMP280 também?🙏

[EmuFlight_0.3.7_RUSHBLADEF7_Build_2019_2088e6f5a.zip](https://github.com/emuflight/EmuFlight/files/6989181/EmuFlight_0.3.7_RUSHBLADEF7_Build_2019_2088e6f5a.zip)

nerdCopter, 2021-08-16:
* @BeauBrewski , please let me  know as soon as RUSHBLADEF7 BMP280 is validated.
* the code for both FC are in a single PR
* i'd like to merge before setting up for 0.4.0 release so that i can "`freeze`" from merging for a bit.

nerdCopter, 2021-08-16:
@BeauBrewski , here is 0.3.4 version for public-testers 
[TESTING_BMP280_EmuFlight_0.3.4_RUSHBLADEF7_Build_52dfc2642.hex.zip](https://github.com/emuflight/EmuFlight/files/6994520/TESTING_BMP280_EmuFlight_0.3.4_RUSHBLADEF7_Build_52dfc2642.hex.zip)

JulioCesarMatias, 2021-08-17:
@BeauBrewski  can you tell if it worked for a RUSHBLADEF7?

BeauBrewski, 2021-08-18:
@JulioCesarMatias I still haven't heard back from the guy who asked for it.

BeauBrewski, 2021-08-22:
Got reply that test file 0.3.4 and new release 0 4.0 did not show the baro for RushBladeF7 and BMP280.

JulioCesarMatias, 2021-08-22:
this is weird.

1 - Is FC looking for the BMP280 automatically on the I2C bus? 

2 - Is the SDA and SCL connection correct?

JulioCesarMatias, 2021-08-22:
> Got reply that test file 0.3.4 and new release 0 4.0 did not show the baro for RushBladeF7 and BMP280.

this is weird.

1 - Is FC looking for the BMP280 automatically on the I2C bus?

2 - Is the SDA and SCL connection correct?

BeauBrewski, 2021-08-22:
1 - I'm not sure but can try to ask.
2 - I am assuming so as it works fine with BF.
Here is screen cap of it not working on Emu w/ the 0.3.4 test build:
[image]

JulioCesarMatias, 2021-08-22:
> 1 - I'm not sure but can try to ask.
> 2 - I am assuming so as it works fine with BF.
> Here is screen cap of it not working on Emu w/ the 0.3.4 test build:
> [image]

I've also seen some boards only work the barometer with the battery connected, I don't know if this is the case

BeauBrewski, 2021-08-23:
Not showing up with battery connected

JulioCesarMatias, 2021-08-24:
> Not showing up with battery connected also.

How sad, I don't know why it didn't work, I did everything correctly in PR #671

JulioCesarMatias, 2021-08-24:
which .hex file did you send the guy to test? Mine or NerdCopter?

JulioCesarMatias, 2021-08-24:
I was downloaded and looked at NerdCopter's .hex file, and the one he uploaded is smaller than mine. Are you sure you sent mine?
[image]

BeauBrewski, 2021-08-24:
I had sent nerdcopters .hex. Did the code make it into 0.4.0 for this fw? The tester also had tried 0.4.0 without baro showing.

nerdCopter, 2021-08-24:
i cherry-picked you mods into older branch, because i did not want the pilot to flash something taht the 0.3.6 GUI was not ready to set all features.   now that 0.4.0 is out, you can merge master  into your branch and send out new .hex.   

however, the cherry-pick used you target-definition, so it's still an unsolved problem irregardless.

JulioCesarMatias, 2021-08-24:
ok @BeauBrewski  and @nerdCopter , so i don't know anymore why it doesn't work.

BeauBrewski, 2021-08-24:
Crap. I had an idea/realization. This has been RushBladeF7 HD flight controller that it is being tested on. I don't know if there is any real difference between the `RUSHBLADEF7` and `RUSHBLADEF7HD`

xpertcrush, 2022-02-04:
I have rush blade f7 hd, if you need me to test something let me know

nerdCopter, 2026-05-07:
resolved by https://github.com/emuflight/EmuFlight/pull/1157

[EmuFlight_0.4.3_RUSHBLADEF7HD_Build_9ac815e.zip](https://github.com/user-attachments/files/27491299/EmuFlight_0.4.3_RUSHBLADEF7HD_Build_9ac815e.zip)