emuflight/EmuFlight #677: No baro  on IFF4 E target

olegshty, 2021-08-18:
**Describe the bug**
 BMP280(I2C) drivers are missing

**To Reproduce**
Turn on baro on configuration tab

**Expected behavior**
baro should turn on

**Flight controller configuration**
'''
# version
# EmuFlight / IFF4_E (IFRC) 0.3.4 Apr 16 2021 / 19:17:34 (1014a9331) MSP API: 1.50

board_name 
manufacturer_id 
mcu_id 00540026425350042034354e
signature 

# name
name * * * *

# resources
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 C09
resource MOTOR 4 C08
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
[... 1085 lines cut ...]
set rate_end_weight = 0
set tpa_rate_p = 75
set tpa_rate_i = 125
set tpa_rate_d = 65
set tpa_breakpoint = 1600
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set vbat_comp_type = OFF
set vbat_comp_ref = 37
set vbat_comp_throttle_level = 75
set vbat_comp_pid_level = 75

# restore original rateprofile selection
rateprofile 0

# save configuration
save
'''
**Setup / Versions**
 - Flight controller [Iflight succex e f4 all versions];
 - Other components [elrs r9mm, vista+nebula pro, all versions];
 - how are the different components wired up: perfecly

**Additional context**
Add any other context about the problem here.

BeauBrewski, 2021-08-18:
Similar to #672 and #415

@JulioCesarMatias @nerdCopter we have another one.

nerdCopter, 2021-08-19:
* will be in 0.4.0 release

closed 2021-08-19 by nerdCopter