emuflight/EmuFlight #684: EmuFlight 0.4 AirMode always on in stab mode?

tricade1, 2021-08-20:
**Describe the bug**
In Stab mode Air Mode is always on

**To Reproduce**
If I fly Acro everything is fine, as soon I switch Stab mode Air mode is on

**Expected behavior**
If I turn Airmode OFF I would assume its off ;)

**Flight controller configuration**
```
# 

# dump

# version
# EmuFlight / BEEBRAIN_BL (BBBL) 0.4.0 Aug 19 2021 / 14:43:24 (a5e2673e4) MSP API: 1.51

board_name 
manufacturer_id 

# name
name AcroBee

# resources
resource BEEPER 1 B10
resource MOTOR 1 B09
resource MOTOR 2 B07
resource MOTOR 3 B06
resource MOTOR 4 B08
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A00
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 NONE
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
[... 739 lines cut ...]
set das_roll_with_yaw_input = 20
set roll_pitch_mag_expo = 0
set rate_center_sensitivity = 100
set rate_end_sensitivity = 125
set rate_center_correction = 10
set rate_end_correction = 75
set rate_center_weight = 0
set rate_end_weight = 40
set tpa_rate_p = 75
set tpa_rate_i = 125
set tpa_rate_d = 65
set tpa_breakpoint = 1750
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set vbat_comp_type = OFF
set vbat_comp_ref = 37
```
**Setup / Versions**
 - Flight controller BEEBRAIN_BL NewBeeDrone Acro Bee 75
[EmuFlight 0.4_AllesOK.txt](https://github.com/emuflight/EmuFlight/files/7021911/EmuFlight.0.4_AllesOK.txt)

 - Other components Crossfire thats all

**Additional context**
If I flash back to Version 0.3.3 with the SAME dump everything is fine ! It appears only in 0.4

Quick-Flash, 2021-08-20:
Do you have airmode set to a switch? We have not intentionally changed how airmode works. I don't think we have any real bugs regarding this, but I could be wrong. Be sure rather than just using the "SAME" dump that you double check 0.4.0 to ensure that no mistakes were made when dumping code between versions.

tricade1, 2021-08-21:
No air mode is disabled in modes. 
i will flash my device again to version 0.3 and 0.4 and use the same dump. Give you an update later.

tricade1, 2021-08-21:
So I installed 0.3.4 and used the dump above - no problems no concerns no airmode :) so looks like something is not right in 0.4

tricade1, 2021-08-22:
Update - I flashed to 0.4 and configured my device with the configurator without using the dump. Now it’s working fine. For some reason importing the 0.3.x dump does not work well

nerdCopter, 2021-08-22:
thank you @tricade1 , the issue was a confusing one. i'm happy it is not a problem anymore :)

closed 2021-08-22 by nerdCopter