emuflight/EmuFlight #690: Xrotor Flight Controller F7 Convertible.

mrpkumano, 2021-08-22:
**Vendor/Model**
* Please provide the target name. e.g. `MAMBAF722`
* Please provide vendor and model of the FC. e.g. `Diatone MAMBA F722 MINI STACK Flight Controller`
* Please provide a link to the vendor specifications.
 e.g https://www.diatoneusa.com/store/p654/New_Diatone_MAMBA_F722_MINI_STACK_Flight_Controller_and_MAMBA_306_ESC.html

**CLI dump or Unified Target**
* Please provide a CLI `dump` of **DEFAULTS** or link to unified target.
  Find the unified targets here: https://github.com/betaflight/unified-targets/tree/master/configs/default

dentex, 2021-08-22:
I was wondering the same, if we have a target for the HOBBYWING-XROTORF7CONV ... but you pasted the example data (MambaF7) ... you should have specified name/model/vendor's link for the XRotor.

I'll try myself:
**Vendor/Model**

- target name: `HOBBYWING_XROTORF7CONV`
- vendor/model: `HOBBYWING XRotor Flight Controller F7 Convertible for FPV Racing`
- link: https://www.hobbywingdirect.com/collections/xrotor-flight-controller-f4/products/xrotor-flight-controller-f7?variant=39331002646643

**Unified Target:**

-  https://github.com/betaflight/unified-targets/blob/master/configs/default/HOWI-HOBBYWING_XROTORF7CONV.config

mrpkumano, 2021-08-22:
ooops!

BeauBrewski, 2021-08-24:
There is already a target request for this with link to test target in the comments.
#625

nerdCopter, 2021-08-24:
`HOWI_HOBBYWING_XROTORF7CONV` was sent to EmuConfigurator for 0.4.0

closed 2021-08-24 by nerdCopter