emuflight/EmuFlight #693: regression: F3 targets bloated: "make" failing

gretel, 2021-08-22:
another round (0.4.0) of some targets not building. `make` should always succeed.
expect the response to be "not supported" but then, references to the targets should be removed.
best approach would be to fix the lack of conditionals coming from feature bloat and lack of testing on older targets.

```
Linking OMNIBUS
Memory region         Used Size  Region Size  %age Used
           FLASH:      238452 B       252 KB     92.41%
    FLASH_CONFIG:          0 GB         4 KB      0.00%
             RAM:       44244 B        40 KB    108.02%
             CCM:          2 KB         8 KB     25.00%
       MEMORY_B1:          0 GB         0 GB
./EmuFlight/tools/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: obj/main/EmuFlight_OMNIBUS.elf section `.bss' will not fit in region `RAM'
./EmuFlight/tools/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: region `RAM' overflowed by 3284 bytes
collect2: error: ld returned 1 exit status
make[1]: *** [obj/main/EmuFlight_OMNIBUS.elf] Error 1
```

```
Linking CRAZYBEEF3FR
Memory region         Used Size  Region Size  %age Used
           FLASH:      242336 B       252 KB     93.91%
    FLASH_CONFIG:          0 GB         4 KB      0.00%
             RAM:       45168 B        40 KB    110.27%
             CCM:          2 KB         8 KB     25.00%
       MEMORY_B1:          0 GB         0 GB
./EmuFlight/tools/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: obj/main/EmuFlight_CRAZYBEEF3FR.elf section `.bss' will not fitin region `RAM'
./EmuFlight/tools/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: region `RAM' overflowed by 4208 bytes
collect2: error: ld returned 1 exit status
make[1]: *** [obj/main/EmuFlight_CRAZYBEEF3FR.elf] Error 1
```

Quick-Flash, 2021-08-22:
We have a special branch that fixes this. Needs some flight testing though.

gretel, 2021-08-22:
> We have a special branch that fixes this. Needs some flight testing though.

would you mind referencing the according commit here?

nerdCopter, 2021-08-22:
`20210215_attempt_bring_F3`
`1343dbdcd` is last 0.3.3 efforts, then master 0.4.0 is merged in.

gretel, 2021-08-24:
an attempt was made... don't get why it was released this way.

closed 2023-11-06 by gretel