emuflight/EmuFlight #694: [Target Request] RUSH_BLADE_F7_HD

BeauBrewski, 2021-08-24:
**Vendor/Model**
* RUSH_BLADE_F7_HD
*Rush FPV Rush Blade F722 Digital
* https://www.racedayquads.com/collections/rush-fpv-products/products/rush-fpv-rush-blade-f722-digital-30x30-flight-controller-for-dji

**CLI dump or Unified Target**
* https://github.com/betaflight/unified-targets/blob/master/configs/default/RUSH-BLADE_F7_HD.config

BeauBrewski, 2021-08-24:
- Need an updated for 0.4.0 test target (only 0.3.3 available). [https://cdn.discordapp.com/attachments/691022009865338950/810631515587215400/EmuFlight_0.3.3_RUSH_BLADE_F7_HD_Build_ae6a0bd2a.bin](url)

- It has been reported that `RUSHBLADEF7` works on this FC, and that the only really difference is the HD plugin on the FC.

- Baro problems with BMP280 (see #672) on RushBlade F7HD flight controller using `RUSHBLADE47`

BeauBrewski, 2021-11-21:
Needs tested!
Check gyro orientation, motor arrangement, flash, barometer, vtx control, everything else if possible please update here
PROPS OFF FOR SAFETY
[EmuFlight_0.4.0_RUSHBLADEF7HD_Build_bc1819e1f.zip](https://github.com/emuflight/EmuFlight/files/7576010/EmuFlight_0.4.0_RUSHBLADEF7HD_Build_bc1819e1f.zip)

AnotherRumor, 2022-07-06:
Has this been confirmed working?

BeauBrewski, 2022-07-07:
> Has this been confirmed working?

Uncertain. Need someone to test and let us know.

AnotherRumor, 2022-07-07:
> > Has this been confirmed working?
> 
> Uncertain. Need someone to test and let us know.

I flashed it yesterday and everything seems good! Had to remap the motor outputs but I'm guessing that's exclusive to my build.

BeauBrewski, 2022-07-29:
Thanks for the feedback, @AnotherRumor 😄

PR #837 (2022-07-29): TARGET RUSH_BLADE_F7_HD
Tested per https://github.com/emuflight/EmuFlight/issues/694
Closes #694
closed 2022-07-29 by nerdCopter