emuflight/EmuFlight #702: MatekF411/Tyro79 - i2c2 works on betaflight but not emuflight

MATH3U5, 2021-09-11:
I working on Tyro79FC , is a Eachine STRM32F411ce flight controller. It is full compatible with the MatekF411 target, but  SDA and SCL pins are not exposed on the board. I would like to add a BMP280 barometer for altitude hold .According to the datasheet I should be able to use the S5 pad from PB03 as i2c2 SDA; and the S6 pad from PB10 as i2c2 SCL.

I know my wiring and port numbers are correct and get sucess in getting the barometer to working on i2c2 using these pins in betaflight with the stock MatekF411 firmware and remaping the pins using betaflight's resource mapping.

I get sucess in betaflight with this commands, and response all ok in status and sensors output:

[betaflight sucess steps for barometer.txt](https://github.com/emuflight/EmuFlight/files/7147048/betaflight.sucess.steps.for.barometer.txt)

resource I2C_SCL 2 B10
resource I2C_SDA 2 B03
set baro_i2c_device = 2
set baro_hardware = BMP280
and get baro for confirm

In Emufligh, barometer does not detect any barometer. Even defining barometer to BMP280. i2c speeds no makes any difference.

I tested all versions of Emuflight until the last one and all firmwares until the last, but all giving me the same result, no baro.

Betaflight with status,resource and dump:
[BTFL_cli_M_20210910_231257.txt]
(https://github.com/emuflight/EmuFlight/files/7147058/BTFL_cli_M_20210910_231257.txt)

Emuflight with status,resource,dump and get baro:

[EMUF_cli_20210910_232249.txt]
(https://github.com/emuflight/EmuFlight/files/7147062/EMUF_cli_20210910_232249.txt)

Thanks in advance for any help you can offer.

Also wanted to say thanks to all person was work so hard to make Emuflight. Is incredible how this is so smooth in my quads, movements and maneuvers is executed with class!

nerdCopter, 2021-09-15:
@MATH3U5 , you could try this... no guarantees.  I hard-coded those resources you listed. if it works, this is **not** really a solution that we can merge to master-code.

[EmuFlight_0.4.0_MATEKF411_Build_uncommitted_20210915CDT__hard_coded_baro.hex.zip](https://github.com/emuflight/EmuFlight/files/7172867/EmuFlight_0.4.0_MATEKF411_Build_uncommitted_20210915CDT__hard_coded_baro.hex.zip)

MATH3U5, 2021-09-20:
Oh, thanks, in this days i learned some of build custom targets, and solved the problem, but i I forgot to send here rsrs, i will upload, but is the same code basicly, no difference, yet anyway thanks, but damn, the emuflight is the best firmware we i use( i am want not fawning over, but is the reality), and have the best results in my quads, and the most static hover, detail, without any GPS, equal a dji phantom or Mavic, but the unique variation is of alt hold, is interessing the implementation of alt hold, I leave this as a suggestion, because the inav is a little complicated os configure and use, is much information.

[EmuFlight_0.4.0_MATEKF411_Build_uncommitted_20210912-03.zip](https://github.com/emuflight/EmuFlight/files/7193927/EmuFlight_0.4.0_MATEKF411_Build_uncommitted_20210912-03.zip)

and annexing target.h because is more useful for community

[MATEKF411-target.h.zip](https://github.com/emuflight/EmuFlight/files/7193962/MATEKF411-target.h.zip)

nerdCopter, 2021-09-20:
@MATH3U5 , Do you think this code is valid for only your issue, or should it be part of master-code?
if the latter, then could you make a pull-request with the modified file?

nerdCopter, 2021-09-20:
i wonder if BOTH I2C1 & i2C2 could be defined at the same time, would this also solve the problem?

@Kaiowarez , any feedback on this?