emuflight/EmuFlight #704: Jeti EXBUS not working on HGLRC Zeus F745, but does on Betaflight

jimmiebfulton, 2021-09-26:
Jeti EXBUS receiver protocol doesn't seem to be working in EmuFlight.

I'm new to EmuFlight, setting it up for the first time.  Therefore, I'm not sure if this is an issue with EmuFlight firmware, or the specific board I'm using.  I suspect it is EmuFlight.  I'm able to flash to Betaflight, and everything works fine there.

* I have an HGLRC Zeus F745 20x20 FC with my receiver wired to TX on UART1 (TX1, which provides telemetry over EXBUS on Betaflight)
* The board is flashed with 0.4.0 with target HGLRCF722.
* I have 'Serial RX' turned on for UART1 in the Ports tab
* On the Configuration tab under Receiver, I have "Serial-based receiver(SPEKSAT, SBUS, SUMD)" selected
* For the Serial Provider, I have "JETIEXBUS" selected.

After Saving, rebooting/repowering, and going to the Receiver tab, I get no receiver output.

With this exact configuration under Betaflight, this all works, and is the standard setup for using Jeti Receivers on Betaflight.

I notice on the Ports tab, the Telemetry Output dropdown does not include Jeti, which seems to indicate to me that remnants of Jeti support from Betaflight are still in EmuFlight, but is no longer working.

jimmiebfulton, 2021-09-26:
I just loaded EmuFlight onto a HolyBro Kakute F7HDV with a working Jeti Receiver setup on Betaflight, and Jeti does not work on EmuFlight on this target, either.  It appears Jeti support is broken in EmuFlight.

nerdCopter, 2022-08-08:
in 2021, i didn't even know where to look..
today i'm noticing that `USE_SERIALRX_JETIEXBUS` & `USE_TELEMETRY_JETIEXBUS` are only defined if FC's flash is >= 256MB.

I wonder if these FC's flash are smaller... 

i could potentially force enable thes for testing purposes.  If it still fails, then something broken is inherited from Butterflight/betaflight 3.5.x

[EmuFlight_0.4.1_Test-builds-for-JETIEXBUS.zip](https://github.com/emuflight/EmuFlight/files/9283461/EmuFlight_0.4.1_Test-builds-for-JETIEXBUS.zip)