emuflight/EmuFlight #720: [Target Request] BETAFPVF411ELRS

BeauBrewski, 2021-11-06:
**Vendor/Model**
BETAFPVF411ELRS

BetaFPV F4 1S 5A AIO Brushless Flight Controller (ELRS 2.4G)
and
BetaFPV  F4 1S 12A AIO FC(ELRS 2.4G V2.0)

https://betafpv.com/collections/brushless-flight-controller/products/f4-1s-5a-aio-brushless-flight-controller-elrs-2-4g
https://betafpv.com/collections/brushless-flight-controller/products/f4-1s-12a-flight-controller

**CLI dump or Unified Target**
https://support.betafpv.com/hc/en-us/articles/4408029285017-CLI-for-F4-1S-AIO-5A-FC-ELRS-2-4G-
https://support.betafpv.com/hc/en-us/articles/4408021512473-CLI-for-F4-1S-AIO-12A-FC-ELRS-2-4G-V2-0-

nerdCopter, 2022-02-01:
no one ported ELRS SPI 
ref: https://github.com/betaflight/betaflight/pull/10788/files

Xatonic, 2022-02-01:
@nerdCopter Do you happen to know if it's being worked on?

BeauBrewski, 2022-02-02:
@Xatonic It's being sporadically worked on, but I haven't made much progress on porting the main SPI ELRS code over to Emu compatibility. That's the main hold up.

nerdCopter, 2022-02-07:
i dont think it will be so easy.  seem they had timing overhaul as a pre-requisite.  also some post-fixes.
https://github.com/betaflight/betaflight/pulls?q=is%3Apr+is%3Amerged+ELRS%2CExpressLRS
maybe this is not all of them as well

nerdCopter, 2022-02-07:
i also asked Phobos, (ELRS original PR), if he would port, but no response.

closed 2024-04-19 by nerdCopter