emuflight/EmuFlight #721: [Target Request] FLYWOOF411EVO_HD

BeauBrewski, 2021-11-06:
**Vendor/Model**
`FLYWOOF411EVO_HD`

Flywoo GOKU GNF4EVO AIO Flight Controller
https://flywoo.tawk.help/article/goku-gnf4evo-aio

**CLI dump or Unified Target**
https://github.com/betaflight/unified-targets/blob/master/configs/default/FLWO-FLYWOOF411EVO_HD.config

Target has been made #718, just needs tested

BeauBrewski, 2021-11-06:
@tstibor can we get a copy of the .hex added to this thread?

tstibor, 2021-11-07:
[EmuFlight_0.4.0_FLYWOOF411HD_Build_d2d689445.zip](https://github.com/emuflight/EmuFlight/files/7491978/EmuFlight_0.4.0_FLYWOOF411HD_Build_d2d689445.zip)

BeauBrewski, 2021-11-07:
> [EmuFlight_0.4.0_FLYWOOF411HD_Build_d2d689445.zip](https://github.com/emuflight/EmuFlight/files/7491978/EmuFlight_0.4.0_FLYWOOF411HD_Build_d2d689445.zip)

Needs tested!
Check gyro orientation, motor arrangement, flash, osd, vtx control, everything else if possible please update here
PROPS OFF FOR SAFETY

BeauBrewski, 2021-11-21:
Reports of great flying so far, with no noticeable missing features according to the one tester Steve_EOD on discord.

closed 2024-04-19 by nerdCopter