emuflight/EmuFlight #722: [Target Request] LDARCF411

BeauBrewski, 2021-11-15:
**Vendor/Model**
LDARCF411
LDARC MR20 F411AIO E35A ESC
https://www.aliexpress.com/item/1005003271826657.html?srcSns=sns_Copy&spreadType=socialShare&bizType=ProductDetail&social_params=60112052035&aff_fcid=c0b4a5e0c7234cc18c6639dd62641dc5-1636987841785-01231-_uR2Mwu&tt=MG&aff_fsk=_uR2Mwu&aff_platform=default&sk=_uR2Mwu&aff_trace_key=c0b4a5e0c7234cc18c6639dd62641dc5-1636987841785-01231-_uR2Mwu&shareId=60112052035&businessType=ProductDetail&platform=AE&terminal_id=1b7f6a1140c04afab6fc8888c9b99689

**CLI dump or Unified Target**
https://github.com/betaflight/unified-targets/blob/master/configs/default/LDAR-LDARC_F411.config

VolkerGoeschl, 2021-11-17:
This should work for the HAKRC version (https://www.aliexpress.com/item/1005001626580846.html) as well. The recommended Target (MatekF411) wont show the Blackbox.

BeauBrewski, 2021-11-17:
> This should work for the HAKRC version (https://www.aliexpress.com/item/1005001626580846.html) as well. The recommended Target (MatekF411) wont show the Blackbox.

That's weird they recommend `MATEKF411` as even the betaflight config does not show blackbox support with that target. And `HAKRC F411` won't work as it has different setup.

BeauBrewski, 2021-11-21:
Needs tested!
Check gyro orientation, motor arrangement, flash, osd, vtx control, everything else if possible please update here
PROPS OFF FOR SAFETY
[EmuFlight_0.4.0_LDARCF411_Build_15baaaca6.zip](https://github.com/emuflight/EmuFlight/files/7576012/EmuFlight_0.4.0_LDARCF411_Build_15baaaca6.zip)

VolkerGoeschl, 2021-11-21:
Thanks for the build! Will test on the HAKRC next week. Maybe the LDARC board will also arrive next week -> will test on that as well!

VolkerGoeschl, 2021-11-23:
Just tested the your build on the HAKRC 35Amp AIO and everything I use (Blackbox, both UARTS gyro orientation, motor mappings) works.
What I haven't tested: buzzer, LED, sbus, soft serial.

BeauBrewski, 2021-11-23:
Thank you, @VolkerGoeschl! How did you even find out that `LDARCF411` works for that board? 😆  Manufacturer recommends `MATEKF411`, but I do not think you would have blackbox if you use that target, otherwise they're same target. Glad it is working for you 😊

BeauBrewski, 2022-03-12:
@VolkerGoeschl 
Any updates on this target and your two boards? Feedback is much appreciated for as many features as possible or available:
✔️/❌ Flash
✔️/❌ Gyro
✔️/❌ Accelerometer
✔️/❌ Motors
✔️/❌ Onboard ADC
✔️/❌ OSD (analog)
✔️/❌ Smart Audio controls (OSD and CLI)
✔️/❌ Alarm switch
✔️/❌ Barometer
✔️/❌ Blackbox
✔️/❌ LED
✔️/❌ GPS
Or any other features or issues you notice

VolkerGoeschl, 2022-03-12:
Sorry for not answering here (did only put a short "lt works as expected" answer in the discord).
I am flying both boards without a problem for a few month now and I can confirm everything except OSD, SmartAudio, LED and GPS (I'm flying DJI + expressLRS so I don't have any free Uarts).

What's "Alarm switch"? If it's the buzzer-pad: I don't use that either but I could test it with a multimeter on the bensh if absolutely needed since it's just floating high getting pulled to ground on activation.

BeauBrewski, 2022-03-12:
I guess I should rephrase that as Beeper instead of Alarm. Don't worry about the buzzer pads testing unless you're just bored. Flying without a problem for a few months is awesome! Thank you for your feedback 😄

PR #838 (2022-07-29): Target LDARC_F411
Test good per https://github.com/emuflight/EmuFlight/issues/722
Closes #722
closed 2022-07-29 by nerdCopter