emuflight/EmuFlight #723: [Target Request] GEPRCF722BT

BeauBrewski, 2021-11-15:
**Vendor/Model**
GEPRCF722BT
GEPRC SPAN-F722HD-BT V1.1 flight controller?
This target/config is different from the GEPRC SPAN F7 BT HD target, which is recommended as EXF722DUAL.

https://github.com/betaflight/unified-targets/blob/master/configs/default/GEPR-GEPRCF722BT.config

BeauBrewski, 2021-11-21:
Needs tested!
Check gyro orientation, motor arrangement, flash, osd, vtx control, everything else if possible please update here
PROPS OFF FOR SAFETY

[EmuFlight_0.4.0_GEPRCF722BT_Build_71d395917.zip](https://github.com/emuflight/EmuFlight/files/7576432/EmuFlight_0.4.0_GEPRCF722BT_Build_71d395917.zip)

hokowi, 2021-11-27:
Tried but the RX is not working - using Happymodel ExpressLRS RX on UART6. Gyro is working but I need to rotate yaw 180 although the board wasn't rotated in my quads. The motor is working fine when I test in Emu configurator. I flashed the same board to target MAMBAF722I2C previously with some resource mapping and it has been working fine. I have attached the CLI dump from both targets [here](https://drive.google.com/drive/folders/1-5VuTrMb9WyZQAxBPlgV7vLjxDJy5EYF?usp=sharing) and hope it helps. Thanks for working on this request.

BeauBrewski, 2021-11-27:
@hokowi Can you get me a DUMP of the `GEPRCF722BT`, you have the DIFF posted and its missing a lot of info.

hokowi, 2021-11-27:
ok I've added a DUMP in the same folder as above

BeauBrewski, 2021-11-28:
@hokowi do you have a screen cap of your ports tab? It looks like you have `SERIAL RX` turned on for  UART2 and UART6. This is your most likely cause for ELRS not working on UART6. Turn off the UART2 `SERIAL RX`.

hokowi, 2021-11-28:
@BeauBrewski you are right - the RX works now after I turned off Serial RX on UART2. I didn't realize that was still on even though I've copied over the old CLI.

BeauBrewski, 2021-11-28:
@hokowi its set as default for that from the manufacturer/betaflight config. I'm glad that is the fix because otherwise everything was looking identical and I was really going to be stumped 😆

PR #839 (2022-08-01): TARGET GEPRCF722BT
Tested per https://github.com/emuflight/EmuFlight/issues/723
Closes #723
closed 2022-08-01 by nerdCopter