emuflight/EmuFlight #733: [TARGET REQUEST] SKYSTARSF7HDPRO

seb3876, 2021-11-25:
Hi,

anyone could send me target for SKYSTAR F7HDPRO?

Thanks a lot!

BeauBrewski, 2021-11-25:
Needs tested!
Check gyro orientation, motor arrangement, flash, osd, vtx control, everything else if possible please update here with feedback that it works or needs improvement.
PROPS OFF FOR SAFETY

[EmuFlight_0.4.0_SKYSTARSF7HDPRO_Build_830c4619e.zip](https://github.com/emuflight/EmuFlight/files/7605533/EmuFlight_0.4.0_SKYSTARSF7HDPRO_Build_830c4619e.zip)

seb3876, 2021-11-26:
Thanks for Target

1/ the connection usb don't run: impossible to connect with computer but via wifi it's ok.
2/ Board and alignment don't ok, i must put -90 to the yaw degrees

BeauBrewski, 2021-11-26:
> 1/ the connection usb don't run: impossible to connect with computer but via wifi it's ok. 2/ Board and alignment don't ok, i must put -90 to the yaw degrees 3/ there are a lot of options missing in the osd

1) I'll look into that. Seem to be having some USB issues with different targets right now. But the target is working and not bricking the boad, so that is good.
2) Alignment is set to the manufacturer recommendations according to BetaFlight configuration. Different builds will have different alignment however.
3) Try turning on "Expert Mode" for more OSD options. It will still have different OSD options than BF however.

seb3876, 2021-11-27:
after many test, baro not recognized.

Could you do something ?

BeauBrewski, 2021-11-27:
What brand/kind of baro? Internal/External?

seb3876, 2021-11-27:
the baro is integrated on fc

BeauBrewski, 2021-11-27:
Did it work before with BF?
Do you have a manufacturer web address for your FC?
Do you have a BF dump of your FC before flashing to Emu?

The SKYSTARS F722HD PRO fc I saw said no baro http://skystars-rc.com/Products/ESC/2021/0620/93.html
[image]

If yours has one, then it must be a different FC but same target, so I will need a copy of that web address and a copy of your DUMP

seb3876, 2021-11-27:
the baro worked on betaflight

this is the dump in betaflight in attached file
[dump all betaflight.txt](https://github.com/emuflight/EmuFlight/files/7611899/dump.all.betaflight.txt)

BeauBrewski, 2021-11-27:
Thank you for that!

BeauBrewski, 2021-12-02:
@seb3876 can I get a copy of your Emuflight dump as well for comparison?

seb3876, 2021-12-02:
hi,

this is the dump in Emuflight in attached file:

[DUMP ALL.txt](https://github.com/emuflight/EmuFlight/files/7643996/DUMP.ALL.txt)

BeauBrewski, 2021-12-02:
and just to clarify, does the barometer light up when connected to speedybee?
[image]

BeauBrewski, 2021-12-02:
Okay, I *think* that helped me fix it. I thought I had all the barometer stuff installed already, but your Emu dump said otherwise. I updated the hex linked above, but am including it again here to be easy. Try it and let me know if it works. 

[EmuFlight_0.4.0_SKYSTARSF7HDPRO_Build_b87117f08.zip](https://github.com/emuflight/EmuFlight/files/7644115/EmuFlight_0.4.0_SKYSTARSF7HDPRO_Build_b87117f08.zip)

seb3876, 2021-12-03:
i tested the new target:
1/ no baro
2/ no information on the osd
3/ always difficult to connect the fc on emuflight

BeauBrewski, 2021-12-03:
@seb3876 Can I get a new Emu Dump of this install please.

BeauBrewski, 2021-12-03:
@seb3876 Thank you again for your patience and help. I spotted my code errors and you may have helped me narrow down the issue with targets connecting poorly with USB. I *think* I fixed OSD, battery, and Baro issues, as before they weren't showing up in the Dump properly, but now they are. 

[EmuFlight_0.4.0_SKSTF7HDPRO_Build_8582a8fbe.zip](https://github.com/emuflight/EmuFlight/files/7650434/EmuFlight_0.4.0_SKSTF7HDPRO_Build_8582a8fbe.zip)

seb3876, 2021-12-05:
@BeauBrewski thank you for your work!

all the previous points are resolved but now the drone turns over as if a propeller was upside down 

can you do something?

thanks you

BeauBrewski, 2021-12-06:
@seb3876 send me a copy of your newest Dump and I'll see if a setting is off. 
In the meantime, double check your board orientation in comparison to the animated drone in the configurator. Double check your props. Double check motor direction. Double check that normal/reversed props setting is correct.

seb3876, 2021-12-06:
@BeauBrewski this is my new DUMP in attached file.
i must put FC -90 to the yaw degrees but quad fly not in good sens
[Dump.txt](https://github.com/emuflight/EmuFlight/files/7661944/Dump.txt)

BeauBrewski, 2021-12-09:
@seb3876 Something is seriously not right with that dump. Looks all glitchy. Either need to try again, or do a reflash with full chip erase.

seb3876, 2021-12-09:
i tried a full erase, new configuration and dump all.

See in attached file.

Thank You

[DUMP ALL EMUFLIGHT.txt](https://github.com/emuflight/EmuFlight/files/7686971/DUMP.ALL.EMUFLIGHT.txt)

seb3876, 2021-12-12:
@BeauBrewski The settings are good.  I don't understand why they don't match.

BeauBrewski, 2021-12-12:
@seb3876 Hmm. I tested on a different target changing the CLI Aux settings and it did not appear to make a difference, so that is not the problem. Sorry about that. It was the only noticeable difference I saw in the dumps. 

I reworked the gyro code and have a new test file.

[EmuFlight_0.4.0_SKSTF7HDPRO_Build_c8cc50dde.zip](https://github.com/emuflight/EmuFlight/files/7699299/EmuFlight_0.4.0_SKSTF7HDPRO_Build_c8cc50dde.zip)

seb3876, 2021-12-14:
@BeauBrewski thank you very much !

BeauBrewski, 2022-03-12:
@seb3876 
Did the newest build ever end up working correctly? I know this was being a headache.
Feedback is much appreciated for as many features as possible or available:
✔️/❌ Flash
✔️/❌ Gyro
✔️/❌ Accelerometer
✔️/❌ Motors
✔️/❌ Onboard ADC
✔️/❌ OSD (analog)
✔️/❌ Smart Audio controls (OSD and CLI)
✔️/❌ Alarm switch
✔️/❌ Barometer
✔️/❌ Blackbox
✔️/❌ LED
✔️/❌ GPS
Or any other features or issues you notice

BeauBrewski, 2022-07-12:
Above target only works with older boards with MPU6xxx gyro.

Will need to add BMI270 support when available.

BeauBrewski, 2023-05-21:
Working on test target
https://github.com/emuflight/EmuFlight/pull/890

BeauBrewski, 2023-05-29:
Needs tested!
Working SKYSTARSF7HDPRO target (name below is shortened for reasons). Bench tested and works on BMI270 flight controllers. Needs flight tested and other features.

[EmuFlight_0.4.2_SKSTRF7HDPRO_BMI270_2b5d8a19c.zip](https://github.com/emuflight/EmuFlight/files/11587141/EmuFlight_0.4.2_SKSTRF7HDPRO_BMI270_2b5d8a19c.zip)

BeauBrewski, 2023-05-29:
**Note: Had USB connect issues--would take 3+ tries to connect into Emu Config. Shortened target name and USB connection is now fast and first try. 
Gyro works-8k/8k pidloop running at 2-4% CPU. Blackbox appears detected.

nerdCopter, 2023-05-29:
does the folder-name need shortening or only the target.h's `USBD_PRODUCT_STRING` ? or both?

nerdCopter, 2023-05-29:
`src/main/pg/board.h`:26:`#define MAX_BOARD_NAME_LENGTH 20`  :thinking:

BeauBrewski, 2023-05-30:
I'm not sure, but I shortened both. SKYSTARSF7HDPRO isn't longer than 20 🤷

BeauBrewski, 2023-05-31:
Update from FB Page:
Everything works great no problem at all !!! but when I enable the led strips feature 2 of my motors stop working
😭

nerdCopter, 2023-05-31:
maybe: TIM_USE_PPM or _ANY, rather than TIM_USE_LED i saw this some place recently

BeauBrewski, 2023-06-05:
> maybe: TIM_USE_PPM or _ANY, rather than TIM_USE_LED i saw this some place recently

That got the motors working, but now LED's don't. 😞

BeauBrewski, 2023-06-05:
Suspecting there might be an issue with LED code/drivers and how it does timers. Tried without drivers, with `light_ws2811strip.c`, and with `light_ws2811strip_hal.c` drivers. Tried `light_ws2811strip_stdperiph.c` but would not compile as its for F4, F3, and F1 boards only.

nerdCopter, 2024-04-19:
target resolved by https://github.com/emuflight/EmuFlight/pull/943 , therefore closing ticket.
any lingering problems unknown at this time.

closed 2024-04-19 by nerdCopter