emuflight/EmuFlight #734: [Target Request] ZEUSF4FR

BeauBrewski, 2021-11-28:
ZEUSF4FR
HGLRC Zeus5 FR AIO 1-2S F411 Flight Controller
https://www.hglrc.com/products/hglrc-zeus5-fr-aio-1-2s-f411-flight-controller-5a-bl_s-4in1-esc-frsky-spi-rx

**CLI dump or Unified Target**
https://github.com/betaflight/unified-targets/blob/master/configs/default/HGLR-ZEUSF4FR.config

BeauBrewski, 2021-11-28:
Needs tested!
Check gyro orientation, motor arrangement, flash, osd, vtx control, everything else if possible please update here with feedback that it works or needs improvement.
PROPS OFF FOR SAFETY

[EmuFlight_0.4.0_ZEUSF4FR_Build_cc98836fe.zip](https://github.com/emuflight/EmuFlight/files/7613747/EmuFlight_0.4.0_ZEUSF4FR_Build_cc98836fe.zip)

BeauBrewski, 2022-03-10:
Vague but good response from one test pilot.

torju01, 2022-05-05:
hello, my zeus aio 5a fr is all ok for me
gyro orientation, flash, osd, vtx control, redpine; all work good !
(i can't confirm the motor arrangement because my build need to turn the aio and so motor remapping)
thank you very much, it is very good to flight with :-)

BeauBrewski, 2022-05-05:
@torju01 are you using it in the Firefly Baby Nano or a custom build?

BeauBrewski, 2022-05-05:
Awesome! Thanks for the reply!

PR #841 (2022-07-29): Target ZEUSF4F4
Tested Per https://github.com/emuflight/EmuFlight/issues/734
Closes #734
closed 2022-07-29 by nerdCopter