emuflight/EmuFlight #735: [Firmware Bug Report] NewBeeDrone BeeBrainPro DSM target gyro not detected.

BeauBrewski, 2021-12-06:
**Describe the bug**
BeeBrainPro DSM target is not showing gyro or accelerometer info when flashing. 

**To Reproduce**
Flash `BEEBRAIN_PRO_DSM` target to appropriate flight controller and connect to configurator. 

**Expected behavior**
Expect Gyro/Accelerometer to be detected.

**Additional context**
NBD's BeeBrainPro DSM target uses a different gyro, `BMI160`, where the EmuFlight target only has `MPU6000`. Need to add if/else code for the correct gyro info.

nerdCopter, 2021-12-06:
this gets the bmi160 gyro working, nothing more: https://github.com/nerdCopter/EmuFlight_nerdRepo/commit/f7c36937cd5d201a847be0a89ae48d2abcdc9d08

BeauBrewski, 2022-04-13:
I'm doing the double define trick like I did with the `MATEKF411RX/CRAZYBEEF4` target. That should keep from losing compatibility with any older boards that still had MPU6000 on the DSM models.

nerdCopter, 2022-04-13:
@BeauBrewski , please do your magic, thank you very much!

nerdCopter, 2024-05-03:
backported to 0.4.2 release assets

closed 2024-05-03 by nerdCopter