emuflight/EmuFlight #736: [Target Request] IFLIGHT_F745_AIO_V2

BeauBrewski, 2021-12-15:
**Vendor/Model**
`IFLIGHT_F745_AIO_V2`
iFlight Beast F7 55A V2 AIO
https://shop.iflight-rc.com/Beast-F7-45A-V2-AIO-MPU6000-25.5-25.5-Pro1625

**CLI dump or Unified Target**
https://github.com/betaflight/unified-targets/blob/master/configs/default/IFRC-IFLIGHT_F745_AIO_V2.config

BeauBrewski, 2021-12-27:
Note, needs driver for BMI270 gyro.

BeauBrewski, 2023-03-14:
Update: It appears newer versions of this flight controller have MPU6000 gyros.
This test target should work with those.
Needs tested!
Check gyro orientation, motor arrangement, flash, osd, vtx control, battery readings, everything else if possible please update here.

PROPS OFF FOR SAFETY!
[EmuFlight_0.4.1_IFLIGHT_F745_AIO_V2_Build_82a2de10b.zip](https://github.com/emuflight/EmuFlight/files/10972167/EmuFlight_0.4.1_IFLIGHT_F745_AIO_V2_Build_82a2de10b.zip)

BeauBrewski, 2023-03-14:
So far, flight, motors, gyro, vtx, and analog osd are working per Moksh on discord channel.

nerdCopter, 2023-05-16:
was this mpu6000 version ever PR'd?

BeauBrewski, 2023-05-17:
Not yet.

BeauBrewski, 2023-05-21:
Working on test target
https://github.com/emuflight/EmuFlight/pull/890

BeauBrewski, 2023-05-21:
Needs tested:

[EmuFlight_0.4.2_IFLIGHT_F745_AIO_V2_Build_719010445.zip](https://github.com/emuflight/EmuFlight/files/11524621/EmuFlight_0.4.2_IFLIGHT_F745_AIO_V2_Build_719010445.zip)

PR #895 (2023-07-12): [Target] IFLIGHT_F745_AIO_V2 (MPU6000 Only)
TARGET - `IFLIGHT_F745_AIO_V2`
MPU6000 ONLY

Tested per issue https://github.com/emuflight/EmuFlight/issues/736

Merge to master if approved.

Do not close #736 as originally BMI270 issue.
closed 2023-07-12 by nerdCopter