emuflight/EmuFlight #744: [Target Request] GEPRC_F722_AIO

salem83, 2021-12-31:
**Vendor/Model**
* Please provide the target name ???
* GEPRC GEP-F722-45A AIO
* https://geprc.com/product/geprc-gep-f722-45a-aio/

**CLI dump or Unified Target**
* [BTFL-GEPRC_F722_AIO-defaults.txt](https://github.com/emuflight/EmuFlight/files/7797203/BTFL-GEPRC_F722_AIO-defaults.txt)
* https://geprc.com/wp-content/uploads/2020/01/CineLog35-HD-TBS-4.2.3-1.txt
* https://github.com/betaflight/unified-targets/blob/master/configs/default/GEPR-GEPRC_F722_AIO.config

BeauBrewski, 2022-01-07:
Needs tested!
Check gyro orientation, motor arrangement, flash, osd, vtx control, everything else if possible please update here with feedback that it works or needs improvement.
PROPS OFF FOR SAFETY

[EmuFlight_0.4.0_GEPRC_F722_AIO_Build_29fdb64ca.zip](https://github.com/emuflight/EmuFlight/files/7826784/EmuFlight_0.4.0_GEPRC_F722_AIO_Build_29fdb64ca.zip)

BeauBrewski, 2023-05-24:
Working on test target
https://github.com/emuflight/EmuFlight/pull/891

BeauBrewski, 2023-05-29:
Needs tested! ICM42688P gyro target.
Works on bench and gyro detected.

[EmuFlight_0.4.2_GEPRC_F722_AIO_icm42x_0c9bb946d.zip](https://github.com/emuflight/EmuFlight/files/11587180/EmuFlight_0.4.2_GEPRC_F722_AIO_icm42x_0c9bb946d.zip)

nerdCopter, 2023-06-29:
@salem83 , have you tested?

nerdCopter, 2024-04-19:
resolved by:
https://github.com/emuflight/EmuFlight/pull/903
https://github.com/emuflight/EmuFlight/pull/1015
closing ticket

closed 2024-04-19 by nerdCopter

salem83, 2024-04-24:
@nerdCopter Sorry, didn't had time back then. And in meanwhile, i switched back to BF.