emuflight/EmuFlight #746: No voltage on Meteor65 (BetaFPV F411)

skrat, 2022-01-17:
I flashed 0.4.0 on my BetaFPV Meteor65 tiny whoop. I used BETAFPV411 target.

The yaw axis on accelerometer seems inverted, but worse, the voltage reading is 0.1 or 0.0 V.

I can't actually take off with it, tried angle mode and then gave up.

Lemme know if you need anything, I'm eager to try Emuflight on this whoop.

skrat, 2022-01-17:
[BTFL_cli_Meteor65_20220116_175512.txt](https://github.com/emuflight/EmuFlight/files/7878800/BTFL_cli_Meteor65_20220116_175512.txt)

This is known working BF conf. I tried assigning the resources in Emuflight, but it led to being unable to connect with USB again, and I had to enter DFU using a physical button and re-flashing.

glebtv, 2022-01-18:
My meteor65 mostly works for me with a MATEKF411RX target (I have frsky spi version)

Here's a diff
[BTFL_cli_Meteor65_20220115_163256.txt](https://github.com/emuflight/EmuFlight/files/7886948/BTFL_cli_Meteor65_20220115_163256.txt)

BUT, I have kinda a lot of rx loss failsafes, i think it was better on betaflight.

glebtv, 2022-01-18:
Voltage works for me, here are the settings

[image]

glebtv, 2022-01-18:
Also, I tried your betaflight settings and somewhy with them i'm missing one ESC.
Here are my betaflight settings
[meteor65_v2.txt](https://github.com/emuflight/EmuFlight/files/7887147/meteor65_v2.txt)

My flight controller is labelled 1S F4 FR VTX v1.1

nerdCopter, 2022-01-18:
* some discord history concurs to use MATEKF411 or MATEKF411RX for SPI version.
* make sure motor `resource`s in CLI are the same from BF dump.
* i also read to find blheli (non-chrome) and reflash ESC's if nothing else works. (i.e. chrome version may introduce error -- i dont know for fact, just passing information along)

skrat, 2022-01-18:
BETAFPVF411 is what I used in BF 4.2.11 (latest stable) and it works flawlessly. ESCs are flashed with Bluejay, and again, works flawlessly in BF. Would @Kaiowarez know anything about this? Also @nerdCopter I see you being co-author of the BETAFPVF411 target. What is this target for then?

nerdCopter, 2022-01-18:
@Kaiowarez may know, or maybe @BeauBrewski could assist.  my only part in the original is that i click the merge buttons, therefore it lists me as co-author, but not really true.

BeauBrewski, 2022-01-19:
@skrat I think you may be using wrong target as your BF dump text says its 
`board_name BETAFPVF411RX`

Try the test file hex for this target:
https://github.com/emuflight/EmuFlight/issues/486#issuecomment-972940387

BeauBrewski, 2022-03-01:
@skrat Any feedback on the test target `BETAFPVF411RX`?

BeauBrewski, 2022-03-13:
Wrong target flashed according to OP's betaflight dump. 
Closing issue. 
Will open new issue if there is problem with correct target.

closed 2022-03-13 by BeauBrewski