emuflight/EmuFlight #750: Unable to set power output with SmartAudio 2.1 VTXs

odddollar, 2022-02-08:
**Describe the bug**
I have a TBS Unify Pro32 Nano connected to a Mamba F405 mini mk2 stack. I am able to use SmartAudio to change channels and bands, but due to the differences between SmartAudio 2.0 and 2.1, I can't change output power. SA 2.0 (which I'm assuming is what Emuflight uses) uses power levels to set VTX output power, such as 0 for pit mode, 1 for the lowest output power, 4 for the highest. This system doesn't work with SA 2.1 devices, which require the dBm value to be sent rather than the power level. With the Unify Pro32 Nano, this would mean 14 being sent as a value to set the VTX to 25mw. Currently the VTX is being sent a value of 1 to set it to this output power. This means that regardless of the power output value I select in the Emuflight OSD, I am unable to change the output power of my VTX above the minimum. Using the button on the VTX only works if it isn't receiving a signal from the FC, so I can disconnect the VTX and set the power manually, but as soon as I reconnect it, Emuflight sets it back to the lowest power output.

**To Reproduce**
Connect a SmartAudio 2.1 device to a Flight Controller flashed with Emuflight. Through the OSD set the power output to the highest level, then disconnect power to the quad. Power the quad again and take note of the flash codes on the VTX; for the Unify Pro32 series, 3 orange flashes followed by 1 blue flash will indicate that it's still on the lowest output power.

**Expected behavior**
The SmartAudio 2.1 VTX to change output powers when different power values are selected in the OSD

**Flight controller configuration**
```
# 

# diff all

# version
# EmuFlight / FURYF4OSD (FY4O) 0.4.0 Aug 23 2021 / 20:40:56 (69caf5a05) MSP API: 1.51

board_name 
manufacturer_id 
mcu_id 003100195453500920373947
signature 

# reset configuration to default settings
defaults nosave

# name
name Problem Child

# resources

# mixer

# servo

# servo mix

# feature

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 2 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 2 1 1300 1700 0 0
aux 2 35 1 1700 2100 0 0
aux 3 0 255 900 900 0 0
aux 4 0 255 900 900 0 0
aux 5 0 255 900 900 0 0
aux 6 0 255 900 900 0 0
aux 7 0 255 900 900 0 0
aux 8 0 255 900 900 0 0
aux 9 0 255 900 900 0 0
aux 10 0 255 900 900 0 0
aux 11 0 255 900 900 0 0
aux 12 0 255 900 900 0 0
aux 13 0 255 900 900 0 0
aux 14 0 255 900 900 0 0
aux 15 0 255 900 900 0 0
aux 16 0 255 900 900 0 0
aux 17 0 255 900 900 0 0
aux 18 0 255 900 900 0 0
aux 19 0 255 900 900 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 15 s 1000

# master
set gyro_lowpass_hz_roll = 210
set gyro_lowpass_hz_pitch = 210
set gyro_lowpass_hz_yaw = 180
set imuf_roll_q = 7000
set imuf_pitch_q = 7000
set imuf_yaw_q = 7000
set imuf_w = 16
set dynamic_gyro_notch_q = 250
set acc_calibration = -84,9,37
set rssi_channel = 16
set rc_smoothing_input_type = BIQUAD
set bat_capacity = 550
set vbat_max_cell_voltage = 42
set vbat_min_cell_voltage = 35
set vbat_warning_cell_voltage = 36
set current_meter = ADC
set ibata_scale = 115
set imu_dcm_kp = 2500
set imu_dcm_ki = 7
set pid_process_denom = 2
set osd_warn_arming_disable = ON
set osd_warn_batt_not_full = OFF
set osd_warn_batt_warning = ON
set osd_warn_batt_critical = ON
set osd_warn_visual_beeper = ON
set osd_warn_crash_flip = ON
set osd_warn_esc_fail = ON
set osd_warn_core_temp = ON
set osd_warn_rc_smoothing = ON
set osd_warn_dji = OFF
set osd_tim1 = 1
set osd_vbat_pos = 449
set osd_rssi_pos = 2490
set osd_tim_1_pos = 2519
set osd_vtx_channel_pos = 429
set osd_crosshairs_pos = 271
set osd_current_pos = 2496
set osd_mah_drawn_pos = 2508
set osd_craft_name_pos = 2081
set osd_warnings_pos = 2442
set osd_avg_cell_voltage_pos = 2465
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set debug_mode = SMARTAUDIO
set vtx_band = 5
set vtx_channel = 3
set vtx_power = 3
set vtx_freq = 5732
set vcd_video_system = PAL

# profile
profile 0

set dterm_lowpass_hz_roll = 220
set dterm_lowpass_hz_pitch = 220
set dterm_lowpass_hz_yaw = 200
set dterm_lowpass2_type = BIQUAD
set dterm_lowpass2_hz_roll = 0
set dterm_lowpass2_hz_pitch = 0
set dterm_lowpass2_hz_yaw = 0
set spa_roll_p = 100
set spa_roll_i = 100
set spa_roll_d = 100
set spa_pitch_p = 100
set spa_pitch_i = 100
set spa_pitch_d = 100
set spa_yaw_p = 100
set spa_yaw_i = 100
set spa_yaw_d = 100
set feathered_pids = 0
set i_decay = 5
set emu_boost = 150
set emu_boost_yaw = 210
set emu_boost_limit = 50
set emu_boost_limit_yaw = 50
set p_pitch = 64
set i_pitch = 60
set d_pitch = 34
set p_roll = 63
set i_roll = 60
set d_roll = 34
set p_yaw = 80
set horizon_tilt_effect = 80
set horizon_strength = 15

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set rates_type = ACTUAL
set roll_rc_rate = 15
set pitch_rc_rate = 15
set yaw_rc_rate = 15
set roll_expo = 54
set pitch_expo = 54
set yaw_expo = 54
set roll_srate = 50
set pitch_srate = 50
set yaw_srate = 40
set tpa_rate_p = 60
set tpa_rate_i = 120
set tpa_rate_d = 70

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```

**Setup / Versions**
 - FC: Mamba F404 Mini MK2
 - VTX: TBS Unify Pro32 Nano
 - Connected together as per [this diagram](https://cdn.shopify.com/s/files/1/0027/2708/4144/files/1_dd74dd94-9608-4df7-b34e-8775883a49f3.jpg?v=1577331449)

nerdCopter, 2022-02-08:
Please update TBS firmware and re-test.
Internal dev said 2.1 is supported and he uses it, but will re-check with Emu 0.4.0 and report. I will comment when received.

nerdCopter, 2022-02-08:
make sure VTX is "Unlocked"

odddollar, 2022-02-08:
Sorry, I should've specified. Yes my VTX is unlocked. I flashed Betaflight onto it temporarily and setup VTX tables and I can change power outputs just fine.

How do I update TBS firmware? Is that the firmware of the VTX?

I'll reflash Emuflight and try again

nerdCopter, 2022-02-09:
sorry, i don't know.  you can join emuflight's discord and ping @madchiller in #help channel. he knows about TBS products. i found this: https://www.team-blacksheep.com/tbs-unify-pro32-ota.pdf

Knightmare5280, 2022-05-19:
I have the same issue but different vtx. I have a rush racing vtx. Same mamba fc. I was able to get POWER OUTPUT AND CHANNEL to change under the tramp protocol but not the s.a.

nerdCopter, 2022-05-19:
@BeauBrewski , any insight to this problem for `rush racing vtx` on `mamba f405`?

BeauBrewski, 2022-05-19:
For both issues, if vtx_low_power_disarm = on, you're going to always show lowest power when disarmed, however OSD should show your assigned power level.

So need to verify that is turned off. 

Another issue if in a the OSD, the vtx is set to Race mode instead of Free mode (not to be mistaken for Fatshark and Race band).  In OSD if you go into Smart Audio and the list of letters/numbers at the top show an F (free mode) at the beginning, the Vtx should be unlocked. If there's an R (race mode), it will lock you in lowest power mode (so you're not blowing everyone out at the race 😆 ). 
[image]

If you go into `CONFIG` from the Smart Audio menu, then select `OP MODEL` and switch from `RACE` to `FREE`

closed 2022-10-18 by odddollar