emuflight/EmuFlight #753: Target Request: SPEEDYBEEF7MINI

BeauBrewski, 2022-02-12:
**Vendor/Model**
`SPEEDYBEEF7MINI`
SpeedyBee F7 Mini 35A 3-6S 8-bit Flight Controller
https://www.speedybee.com/speedybee-f7-mini-35a-3-6s-8-bit-flight-controller-stack/

**CLI dump or Unified Target**
https://github.com/betaflight/unified-targets/blob/master/configs/default/SPBE-SPEEDYBEEF7MINI.config

BeauBrewski, 2022-02-26:
Needs tested!
Check gyro orientation, motor arrangement, flash, osd, vtx control, *BLUETOOTH*, everything else if possible please update here with feedback that it works or needs improvement.
PROPS OFF FOR SAFETY

[EmuFlight_0.4.0_SPEEDYBEEF7MINI_Build_afb7ca3b2.zip](https://github.com/emuflight/EmuFlight/files/8147472/EmuFlight_0.4.0_SPEEDYBEEF7MINI_Build_afb7ca3b2.zip)

Note: May have slow connection issues on USB. Need feedback if this is an issue.

baboon9999, 2022-09-28:
Bit of a late reply from me, but I have this fc and will test on version 0.4.0

nerdCopter, 2022-09-29:
@baboon9999 , please report working conditions for any/all of the following
✔️/❌ Flash
✔️/❌ Gyro
✔️/❌ Accelerometer
✔️/❌ Motors
✔️/❌ Onboard ADC
✔️/❌ OSD (analog)
✔️/❌ Smart Audio controls (OSD and CLI)
✔️/❌ Alarm switch
✔️/❌ Barometer
✔️/❌ BlueTooth
✔️/❌ Blackbox
✔️/❌ LED
✔️/❌ GPS

baboon9999, 2022-10-02:
Some feedback.
✔️Flash
✔️Gyro 
✔️Accelerometer
✔️ Onboard
OSD Not Tested as on DJI but puts our voltage to goggles.
Slight oscillations but thinks it needs tweak.

nerdCopter, 2022-10-02:
thank you. correct, tuning is a pilot chore separate from target-definition.

PR #1041 (2024-04-19): [target] add SPEEDYBEEF7MINI
- resolves #753
closed 2024-04-19 by nerdCopter