emuflight/EmuFlight #763: Target Request: FLYWOOF745NANO

BeauBrewski, 2022-03-04:
**Vendor/Model**
`FLYWOOF745NANO`
GOKU F745 FC 16mm X 16mm
https://flywoo.net/products/goku-f745-fc-16mm-x-16mm-w-4pcs-ws2812-led

**CLI dump or Unified Target**
https://github.com/betaflight/unified-targets/blob/master/configs/default/FLWO-FLYWOOF745NANO.config

BeauBrewski, 2022-03-04:
Needs tested!
Check gyro orientation, motor arrangement, flash, osd, vtx control, everything else if possible please update here with feedback that it works or needs improvement.
PROPS OFF FOR SAFETY

[EmuFlight_0.4.0_FLYWOOF745NANO_Build_da2c7faf5.zip](https://github.com/emuflight/EmuFlight/files/8182746/EmuFlight_0.4.0_FLYWOOF745NANO_Build_da2c7faf5.zip)

crowwingfpv, 2022-08-31:
So far it all works, OSD overlay with DJI works, GPS, flight, etc.. All good so far.

PR #1042 (2024-04-19): [target] add FLYWOOF745NANO
- resolves #763
closed 2024-04-19 by nerdCopter