emuflight/EmuFlight #765: TARGET REQUEST: IFLIGHT_BLITZ_F722

BeauBrewski, 2022-03-06:
**Vendor/Model**
IFLIGHT_BLITZ_F722
IFlight Blitz F7 Flight Controller
https://shop.iflight-rc.com/quad-parts-cat20/electronics-cat27/BLITZ/BLITZ-F7-Flight-Controller-Pro1692

**CLI dump or Unified Target**
https://github.com/betaflight/unified-targets/blob/master/configs/default/IFRC-IFLIGHT_BLITZ_F722.config

BeauBrewski, 2022-03-06:
Needs tested!
Check gyro orientation, motor arrangement, flash, osd, vtx control, everything else if possible please update here
PROPS OFF FOR SAFETY

etsme, 2022-06-12:
Has there been any further testing on this FC ?

BeauBrewski, 2022-06-12:
@etsme none as of yet. If you have a flight controller with this target, you are welcome to test it and give feed back. That is how updates get made on new targets.

BeauBrewski, 2023-05-21:
Working on test target
https://github.com/emuflight/EmuFlight/pull/890

PR #1043 (2024-04-19): [target] add IFLIGHT_BLITZ_F722
- resolves #765
closed 2024-04-19 by nerdCopter