emuflight/EmuFlight #766: GEPRC F411 35A AIO (STM32F411)

MaikFPV, 2022-03-07:
Vendor/Model
GEP-F411-35A AIO
GEPRC Flight Controller F4 35A AIO e.g. Cinelog 30
https://geprc.com/product/gep-f411-35a-aio-f411-fc-35a-2-6s-8bits-bls-esc-25-5mm-m2/

CLI dump or Unified Target
https://github.com/betaflight/unified-targets/blob/master/configs/default/GEPR-GEPRCF411_AIO.config

I tried GEPRCF411_PRO but motor map is not correct.

BeauBrewski, 2022-03-10:
Needs tested!
Check gyro orientation, motor arrangement, flash, osd, vtx control, everything else if possible please update here

PROPS OFF FOR SAFETY!
[EmuFlight_0.4.0_GEPRCF411_AIO_Build_e275b128d.zip](https://github.com/emuflight/EmuFlight/files/8223718/EmuFlight_0.4.0_GEPRCF411_AIO_Build_e275b128d.zip)

MaikFPV, 2022-03-11:
Tested it and it works - UART 1 is twice in settings?! Will test more this weekend. Thanks you BeauBrewski
[image]

BeauBrewski, 2022-03-11:
Hahahaha, I think I know what the issue is, but let me double check with @Kaiowarez.

BeauBrewski, 2022-03-11:
@MaikFPV I had a brain fart when setting up the softserial pins. The target above has been updated, if you want to reflash and let me know if second UART1 is gone. Thank you!

MaikFPV, 2022-03-11:
Fixed! Thanks a lot!
[image]

BeauBrewski, 2022-03-11:
Awesome! Feedback is much appreciated for as many features as possible or available:
✔️/❌ Flash
✔️/❌ Gyro
✔️/❌ Accelerometer
✔️/❌ Motors
✔️/❌ Onboard ADC
✔️/❌ OSD (analog)
✔️/❌ Smart Audio controls (OSD and CLI)
✔️/❌ Alarm switch
✔️/❌ Barometer
✔️/❌ Blackbox
✔️/❌ LED
✔️/❌ GPS
Or any other features or issues you notice

nerdCopter, 2022-03-11:
:)
i really like the ✔️/❌ format if pilots can feedback like so, or at least `Y`/`N` if emojis are a problem.

closed 2022-07-29 by nerdCopter

Shreyash-Patil470, 2023-08-13:
I have bricked my GEPRC F411 AIO, as in when I plug it into a PC I get a red constant light. How do I fix it?

nerdCopter, 2023-08-13:
not bricked. press and hold boot-button, then plug in USB.  you will gain DFU for re-flashing.

PR #844 (2022-07-29): TARGET GEPRCF411_PRO
Tested per https://github.com/emuflight/EmuFlight/issues/766
Closes #766