emuflight/EmuFlight #776: Matek F722HD Duplicated UART1

mdrjr, 2022-03-17:
**Describe the bug**
Did a clean flash, full chip erase, a few times, rolled back to Betaflight and flashed emu back.
Same thing.
It shows two UART1 on Configurator and if you try to assign anything to this uart after the reboot it resets back.
diff is irreleavant since its a clean flash, no changes, however on dump you can see two instances of the serial 0

**To Reproduce**
Just flash emuflight to a Matek F722HD no changes, you have two UART1 on Configurator

**Flight controller configuration**
```
# dump

# version
# EmuFlight / MATEKF722HD (MTKS) 0.4.0 Aug 23 2021 / 20:46:32 (69caf5a05) MSP API: 1.51

board_name 
manufacturer_id 

# name
name -

# resources
resource BEEPER 1 C13
resource MOTOR 1 C08
resource MOTOR 2 C09
resource MOTOR 3 B04
resource MOTOR 4 B05
resource MOTOR 5 B00
resource MOTOR 6 B01
resource MOTOR 7 B10
resource MOTOR 8 B11
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 C10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 C11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource I2C_SCL 1 B08
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
resource I2C_SCL 4 NONE
resource I2C_SDA 1 B09
resource I2C_SDA 2 NONE
resource I2C_SDA 3 NONE
resource I2C_SDA 4 NONE
resource LED 1 A14
resource LED 2 A13
resource LED 3 NONE
resource RX_BIND 1 NONE
resource RX_BIND_PLUG 1 NONE
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 NONE
resource SPI_SCK 4 NONE
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 NONE
resource SPI_MISO 4 NONE
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 C03
resource SPI_MOSI 3 NONE
resource SPI_MOSI 4 NONE
resource ESCSERIAL 1 NONE
resource CAMERA_CONTROL 1 B15
resource ADC_BATT 1 C02
resource ADC_RSSI 1 C00
resource ADC_CURR 1 C01
resource ADC_EXT 1 A04
resource BARO_CS 1 NONE
resource COMPASS_CS 1 NONE
resource PINIO 1 A15
resource PINIO 2 B03
resource PINIO 3 NONE
resource PINIO 4 NONE
resource USB_MSC_PIN 1 NONE
resource FLASH_CS 1 B12
resource SPI_PREINIT_IPU 1 B02
resource SPI_PREINIT_IPU 2 B12
resource SPI_PREINIT_IPU 3 NONE
resource SPI_PREINIT_IPU 4 NONE
resource SPI_PREINIT_IPU 5 NONE
resource SPI_PREINIT_IPU 6 NONE
resource SPI_PREINIT_IPU 7 NONE
resource SPI_PREINIT_IPU 8 NONE
resource SPI_PREINIT_IPU 9 NONE
resource SPI_PREINIT_IPU 10 NONE
resource SPI_PREINIT_IPU 11 NONE
resource SPI_PREINIT_OPU 1 NONE
resource SPI_PREINIT_OPU 2 NONE

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 100 -1
servo 1 1000 2000 1500 100 -1
servo 2 1000 2000 1500 100 -1
servo 3 1000 2000 1500 100 -1
servo 4 1000 2000 1500 100 -1
servo 5 1000 2000 1500 100 -1
servo 6 1000 2000 1500 100 -1
servo 7 1000 2000 1500 100 -1

# servo mix
smix reset

# feature
feature -RX_PPM
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -RANGEFINDER
feature -TELEMETRY
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -OSD
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -RX_SPI
feature -SOFTSPI
feature -ESC_SENSOR
feature -DYNAMIC_FILTER
feature -LEGACY_SA_SUPPORT
feature RX_SERIAL
feature TELEMETRY
feature OSD
feature AIRMODE
feature DYNAMIC_FILTER

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
beeper DISARM_REPEAT
beeper ARMED
beeper SYSTEM_INIT
beeper ON_USB
beeper BLACKBOX_ERASE
beeper CRASH FLIP
beeper CAM_CONNECTION_OPEN
beeper CAM_CONNECTION_CLOSED
beeper RC_SMOOTHING_INIT_FAIL

# beacon
beacon -RX_LOST
beacon -RX_SET

# map
map AETR1234

# serial
serial 20 1 115200 57600 0 115200
serial 0 0 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
serial 3 0 115200 57600 0 115200
serial 4 0 115200 57600 0 115200
serial 5 0 115200 57600 0 115200
serial 0 0 115200 57600 0 115200
```

**Setup / Versions**
 - Matek F722 HD (Aliexpress Matek Store)
 - Matek GPS, Crossfire RX, DJI Air Unit

BeauBrewski, 2022-03-17:
Looks like there's a Soft Serial that got named wrong in the target file so it shows up like a second UART1 instead of a SoftSerial1.
Will get you a fixed target to test in a bit.

BeauBrewski, 2022-03-17:
Give this a test and see if it works. 

[EmuFlight_0.4.0_MATEKF722HD_Build_7d7e81063.zip](https://github.com/emuflight/EmuFlight/files/8295321/EmuFlight_0.4.0_MATEKF722HD_Build_7d7e81063.zip)

mdrjr, 2022-03-18:
Hi @BeauBrewski  nope, it still have two UART1.

BeauBrewski, 2022-03-18:
I just realized that a spare FC I have but haven't put in a build yet uses this target. I'll do some more thorough testing then and get this working. 
Can confirm, extra UART1. 😆

BeauBrewski, 2022-03-18:
And check out whats over on the BF config for this:
[image]

BeauBrewski, 2022-03-18:
Alright, try this target out. I tested it on my own FC and no extra UART1 now. Let us know if there is any other issues.

[EmuFlight_0.4.0_MATEKF722HD_Build_f2f37b8ad.zip](https://github.com/emuflight/EmuFlight/files/8301517/EmuFlight_0.4.0_MATEKF722HD_Build_f2f37b8ad.zip)

mdrjr, 2022-03-19:
@BeauBrewski looks all ok now, flash this version, no duplicates, I could set UART1 to GPS as intended here, hovered without issues.

BeauBrewski, 2022-04-01:
Merged to master. Closing issue.

closed 2022-04-01 by BeauBrewski