emuflight/EmuFlight #777: [Target Request] ZEUSF4EVO

BeauBrewski, 2022-03-17:
**Vendor/Model**
`ZEUSF4EVO`
HGLRC ZEUS10 AIO FPV Racing Drone 2-6S F411 Flight Controller 10A BL_S 4in1 ESC
https://www.hglrc.com/products/hglrc-zeus10-aio-fpv-racing-drone-2-6s-f411-flight-controller-10a-bl_s-4in1-esc

**CLI dump or Unified Target**
https://github.com/betaflight/unified-targets/blob/master/configs/default/HGLR-ZEUSF4EVO.config

BeauBrewski, 2022-03-17:
Needs tested!
Check gyro orientation, motor arrangement, flash, osd, vtx control, everything else if possible please update here

PROPS OFF FOR SAFETY!

[EmuFlight_0.4.0_ZEUSF4EVO_Build_46b5fdb64.zip](https://github.com/emuflight/EmuFlight/files/8287909/EmuFlight_0.4.0_ZEUSF4EVO_Build_46b5fdb64.zip)

zsoltm, 2022-05-18:
Just flashed this firmware, I'm experiencing 80-100% CPU load on the controller.
Apart from that most features seems to be working.

nerdCopter, 2022-05-18:
80-100% doesn't sound right.  try 4k pidloop though; don't want a cpu spike to 100% in-flight.

zsoltm, 2022-05-20:
@nerdCopter totally missed that EMU has 8k/8k for default. Thanks for your input.

With 4k pid loop it has CPU load at around 20-30% (bit higher than BF but that's normal I assume, given EMU does more complex calculations in the loop). It can do 8k/8k at around 80-90% with GPS disabled -- I have caddx vista on this model, so I had to use soft-serial for GPS.

So my evaluation so far:

✔️ Flash
✔️ Gyro
✔️ Accelerometer
✔️ Motors
✔️ Onboard ADC
✔️ Blackbox

Not tested:

? OSD (analog) - no vtx
? Smart Audio controls (OSD and CLI) 
? Alarm switch
? Barometer - this board has no baro
? LED - no leds
? GPS - not installed yet

I plan to test GPS with this model and FC at some point.

nerdCopter, 2022-05-20:
Still weird, my un-soldered FC does about 8% on the bench at 8k.  TYVM
maybe @BeauBrewski will have some advice or resource mods.

BeauBrewski, 2022-05-20:
I know a lot of F411 fc require the gyro/pid loops to be lowered when Emu went to 0.4.0 because CPU load runs too high. I've noticed this more on my Mobulas. I'm not sure what all else can be adjusted to help lower CPU load.

BeauBrewski, 2022-05-20:
Project Mockingbird did some voodoo with setup so that f411 aio could run 8k loops on 0.4.0, but I'm not sure what all was involved in lowering the CPU load. Also had issues with core temp alarms going off after, so I didn't follow all his settings.

nerdCopter, 2022-05-20:
> Project Mockingbird did some voodoo with setup so that f411 aio could run 8k loops on 0.4.0, but I'm not sure what all was involved in lowering the CPU load. Also had issues with core temp alarms going off after, so I didn't follow all his settings.

usually disable dynamic_filter and use only static LPF's

zsoltm, 2022-05-20:
Did a full re-test, with clean re-flash, it looks like communications with serial RX increases the load significantly.

| Use case                             | CPU load* | Details                             |
|--------------------------------------|-----------|-------------------------------------|
| default after clean flash full erase | 44%       | [tasks / status][tdd]               |
| + serial rx                          | 70%       | [tasks / diff / status][td-srx]     |
| + serial rx + telemetry              | 84%       | [tasks / diff / status][td-srx-tel] |

(*) according to the status bar of EMU Configurator

Are these load increases normal just for the additional serial comms with the RX? I think it shoildn't eat up that much CPU. I have ELRS RX with v2.x firmware, connected to hard UART2.

Since the status cmdline command doesn't include serial, it must be noted that the load that is reported by the configurator at the bottom is much higher once serial rx is enabled.

[tdd]: https://gist.github.com/zsoltm/e4eb492c051e629a8b3b4b00591fb911#file-clean-flash-w-full-erase
[td-srx-tel]: https://gist.github.com/zsoltm/e4eb492c051e629a8b3b4b00591fb911#file-plus-serial-rx
[td-srx]: https://gist.github.com/zsoltm/e4eb492c051e629a8b3b4b00591fb911#file-plus-serial-rx-with-telemetry

nerdCopter, 2022-05-20:
good find. consider modifying rc-smoothing settings.  try `interpolation 21` versus `filter PT(n)`

zsoltm, 2022-05-20:
@nerdCopter that is the solution.

I had no idea that ELRS can generate such a high load with its high packet rate, but totally makes sense. I had 500Hz, it's too much for this poor FC even with interpolation. With 250Hz, CPU is 50% with filter, so interpolation vs filter doesn't make such a big difference, at least alone.

nerdCopter, 2022-05-20:
thank you; you have education me.

PR #845 (2022-07-29): TARGET ZEUSF4EVO
Tested per https://github.com/emuflight/EmuFlight/issues/777
Closes #777
closed 2022-07-29 by nerdCopter