emuflight/EmuFlight #793: Target Request: AOCODAF722MINI

BeauBrewski, 2022-04-07:
**Vendor/Model**
`AOCODAF722MINI`
Aocoda RC F7 Mini Flight Controller
https://www.amazon.com/gp/product/B09P617JZ9/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

**CLI dump or Unified Target**
https://github.com/betaflight/unified-targets/blob/master/configs/default/SJET-AOCODAF722MINI.config

BeauBrewski, 2022-04-07:
Needs tested!
Check gyro orientation, motor arrangement, flash, osd, vtx control, everything else if possible please update here

PROPS OFF FOR SAFETY!

[EmuFlight_0.4.0_AOCODAF722MINI_Build_37a309b7d.zip](https://github.com/emuflight/EmuFlight/files/8438986/EmuFlight_0.4.0_AOCODAF722MINI_Build_37a309b7d.zip)

BeauBrewski, 2022-04-07:
@SkylionFPV 
Feedback is much appreciated for as many features as possible or available:
✔️/❌ Flash
✔️/❌ Gyro
✔️/❌ Accelerometer
✔️/❌ Motors
✔️/❌ Onboard ADC
✔️/❌ OSD (analog)
✔️/❌ Smart Audio controls (OSD and CLI)
✔️/❌ Alarm switch
✔️/❌ Barometer
✔️/❌ Blackbox
✔️/❌ LED
✔️/❌ GPS
Or any other features or issues you notice

Ultraexpert, 2022-09-10:
Hey did you manage a working hex for inav 5.1?
Thanks

BeauBrewski, 2022-09-10:
Wasn't planning to. I can't even get verification that this works for Emuflight 🤣

SkylionFPV, 2022-09-10:
I thought I told you that it was working on emu back when I tested it,
sorry I've been MIA for so long my babies and my 3d printers have been
taking up all of my time, but yes the aocoda f7 target works get for me so
far though I haven't flown anything in over a month

BeauBrewski, 2022-09-10:
Ooops, I need to keep better track of discord/fb messenger/etc feedback.
I'll do a 0.4.1 pr and get this merged to master soon then.

@Ultraexpert sorry, I don't work on INAV targets. You're welcome to try out Emuflight, but the GPS functionality is going to be severely lacking compared to INAV. However the flight feel for freestyle and cinematic flight is excellent.

Ultraexpert, 2022-09-10:
> Ooops, I need to keep better track of discord/fb messenger/etc feedback. I'll do a 0.4.1 pr and get this merged to master soon then.
>
> @Ultraexpert sorry, I don't work on INAV targets. You're welcome [...]

Hey thanks I will try emuflight ! Looking forward to that merge :)

Ultraexpert, 2022-09-11:
This this for a t tail fixed wing. Is emuflight ok for this?

closed 2022-09-12 by nerdCopter

nerdCopter, 2022-09-12:
uploaded the 0.4.1 hex to EmuConfigurator.

@Ultraexpert EmuFlight was forked from Butterflight which was forked from Betaflight 3.5.X.
Fixed wing should work, but Emu has older GPS logic (whatever Betaflight 3.5.x had).

Ultraexpert, 2022-09-12:
> uploaded the 0.4.1 hex to EmuConfigurator.
> 
> @Ultraexpert EmuFlight was forked from Butterflight which was forked from Betaflight 3.5.X. Fixed wing should work, but Emu has older GPS logic (whatever Betaflight 3.5.x had).

Ok amazing thanks for the info.