emuflight/EmuFlight #794: Mobula6 V2.0 fails to arm - new hardware?

simeonpashley, 2022-04-08:
**Describe the bug**
Using 0.4.0 firmware, a newly purchased Mobula6 Frsky returns `NO_GYRO` under Setup > Arming Disable Flags and will not arm. The device was purchased on 3/Apr/2022 and now comes with BF4.3 pre-installed.

I have tried a few failed config attempts such as `set acc_hardware = ICM20689` to no avail. I have since re-flashed to Emuflight defaults and now back to BF4.3

This may be unrelated but the [HappyModel website](https://www.happymodel.cn/index.php/2022/02/14/mobula6-v2-0-upgraded-flight-controller-crazybeef4fr-lite-crazybeef4fs-lite-v3-1-version-update-log/) implies a hardware change.

> **Mobula6 V2.0 upgraded flight controller CrazybeeF4FR lite / CrazybeeF4FS lite V3.1**
> The flight controller for Mobula6 and Mobula6 HD Frsky version and Flysky version would upgrade to V3.1 version since 14th .Feb 2022
> Update log:
> MPU6000 changed to ICM20689 because of MPU6000 discontinued
> Onboard VTX changed to OPENVTX support 0mw~400mw
> Add “GND” solder pad for VTX antenna
> Only support Betaflight firmware 4.1.Xversion and latest than 4.1.X not include Legacy target.

**To Reproduce**
1. Flash "new" Mobula6 Frsky with 0.4.0
2. In Configurator, navigate to "Setup"
3.  "Arming Disable Flags" contains "NO_GYRO"

Device will not arm, as expected due to the flag

**Expected behavior**
1. Flash "new" Mobula6 Frsky with 0.4.0

The device Arms.

**Flight controller configuration**

This is newly flashed stock EmuFlight defaults

```
Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# diff all

# version
# EmuFlight / CRAZYBEEF4FR (C4FR) 0.3.4 Apr 16 2021 / 19:23:25 (1014a9331) MSP API: 1.50

board_name 
manufacturer_id 
mcu_id 003f00313439511336383737
signature 

# reset configuration to default settings
defaults nosave

# name

# resources

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# master

# profile
profile 0

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

**Setup / Versions**
Stock Mobula6 Frsky, purchase 3/Apr/2022
Happymodel website implies it's `CrazybeeF4FR lite`

nerdCopter, 2022-04-08:
it seems a new target will have to be defined due to the new ICM20689.
@BeauBrewski , are you up to this? if yes, may have to be creative in target naming.

nerdCopter, 2022-04-08:
@simeonpashley , please provide a **betaflight** `dump` of defaults.  the emuflight diff is unhelpful.
```
defaults nosave
dump all
```
thank you

simeonpashley, 2022-04-08:
When reverting from EmuFlight => BF4.3, I re-installed the defaults from the vendor website.

To save a click, I've copied the frsky version from the site below.

http://www.happymodel.cn/index.php/2022/02/14/default-factory-firmware-and-diff-settings-for-mobula6-v2-0-which-comes-with-new-crazybeef4fr-lite-v3-1-and-crazybeef4fs-lite-v3-1-flight-controller/

```
# feature
feature -TELEMETRY

# serial
serial 1 2048 115200 57600 0 115200

# map
map TAER1234

# aux
aux 0 0 0 1800 2100 0 0
aux 1 1 1 900 1200 0 0
aux 2 35 2 1800 2100 0 0

vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 LOWRACE  L FACTORY 5333 5373 5413 5453 5493 5533 5573 5613
vtxtable powerlevels 5
vtxtable powervalues 10 2 14 20 26
vtxtable powerlabels 0 RCE 25 100 400

# master
set acc_calibration = -304,-58,-200,1
set rx_spi_protocol = FRSKY_D
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set align_board_yaw = 0
set vbat_min_cell_voltage = 300
set vbat_warning_cell_voltage = 310
set yaw_motors_reversed = ON
set small_angle = 180
set osd_warn_batt_not_full = OFF
set osd_warn_batt_warning = OFF
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_rssi_pos = 314
set osd_tim_2_pos = 2433
set osd_flymode_pos = 2457
set osd_throttle_pos = 2425
set osd_vtx_channel_pos = 2101
set osd_current_pos = 2336
set osd_mah_drawn_pos = 2368
set osd_craft_name_pos = 2442
set osd_warnings_pos = 2345
set osd_avg_cell_voltage_pos = 2401
set osd_disarmed_pos = 2314
set osd_rssi_pos = 2082
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_band = 5
set vtx_channel = 6
set vtx_power = 5
set vtx_low_power_disarm = ON
set vtx_freq = 5843
set name = Mobula6 
set pid_process_denom = 4
set vbat_min_cell_voltage = 290
set vbat_warning_cell_voltage = 300
set ibata_scale = 1175

profile 0

# profile 0
set iterm_relax = RPY
set p_pitch = 80
set i_pitch = 100
set d_pitch = 85
set f_pitch = 140
set p_roll = 85
set i_roll = 100
set d_roll = 85
set f_roll = 140
set p_yaw = 120
set i_yaw = 100
set f_yaw = 140
set feedforward_transition = 21

rateprofile 0

# end the command batch
batch end
save
```

nerdCopter, 2022-04-08:
@simeonpashley , Here is my feeble attempt at a target.  No promises; it's probably wrong and we can hope Beau has time for the task.
[EmuFlight_0.4.0_CRAZYBEEF4FRV31_Build_827c1b636.hex.zip](https://github.com/emuflight/EmuFlight/files/8454589/EmuFlight_0.4.0_CRAZYBEEF4FRV31_Build_827c1b636.hex.zip)

@BeauBrewski , this is probably the wrong way, but i copied existing definitions to a new folder, renamed targets and replaced defines for ICM20689 : https://github.com/nerdCopter/EmuFlight_nerdRepo/commit/25a0005de5c29a0a73274693c83e707b80266afb

nerdCopter, 2022-04-08:
> [...] To save a click, I've copied the frsky version from the site below. [...]

but those things don't include resources and definitions.  `dump all` is typical needed.

simeonpashley, 2022-04-08:
A fresh `dump all` via `Configurator: 10.8.0 (340087a4)`

```
# version
# Betaflight / CRAZYBEEF4FR (C4FR) 4.3.0 Feb 12 2022 / 09:44:54 (norevision) MSP API: 1.44

# start the command batch
batch start

board_name CRAZYBEEF4FR
manufacturer_id 
mcu_id 003f00313439511336383737
signature 

# name: Mobula6

# resources
resource BEEPER 1 C15
resource MOTOR 1 B10
resource MOTOR 2 B06
resource MOTOR 3 B07
resource MOTOR 4 B08
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 A02
resource PWM 2 A09
resource PWM 3 A10
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A00
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 NONE
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource LED 1 C13
resource LED 2 NONE
resource LED 3 NONE
resource RX_BIND 1 NONE
[... 1340 lines cut ...]
set yaw_rc_rate = 7
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 67
set tpa_rate = 65
set tpa_breakpoint = 1350
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998
set roll_level_expo = 0
set pitch_level_expo = 0

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```

BeauBrewski, 2022-04-09:
@nerdCopter I should have to add additional gyro info to the `CRAZYBEEF4FR/FS` section of the `MATEKF411RX` target so it can work no matter the gyro. I can do that right away.

BeauBrewski, 2022-04-09:
@simeonpashley 

Give this a quick flash if you don't mind and see if Gyro shows and works for you. 😃 

[EmuFlight_0.4.0_CRAZYBEEF4FR_Build_7d9c91540.zip](https://github.com/emuflight/EmuFlight/files/8457026/EmuFlight_0.4.0_CRAZYBEEF4FR_Build_7d9c91540.zip)

simeonpashley, 2022-04-09:
Confirmed working! Thank you so much! 😍

- Installed `.hex` firmware you supplied
- Used the Pid preset for 0802
- Setup my receiver, rates, modes, etc.

Armed! 
Off it flew! 🚀

Happy mobula6 pilot  :)

BeauBrewski, 2022-04-10:
Do you have the Smart Audio set up correctly on the ports tab on the correct UART?  Should be UART2. If connected right, any of your changes in CLI should show up in the goggles when you access the OSD menu, as well as changes in OSD should be possible and saved.

Also, looking at Happy Model website for the V2 Mob6, and your dump from above, 
`vtx_power = 1` is RCE, which is your PIT MODE in OpenVtx and can cause the symptoms you describe above.
`vtx_power = 2` is 25mW

Try inputting in CLI:
```
set vtx_power=2
save
```
Or any of the powers 2-5. See if that helps.

simeonpashley, 2022-04-10:
Thanks. I’ve tried vtx_power 1-4 and nothing changed. Changing it via the osd didn’t change anything either. The vtx was only visible when the quad was in my hand and I was “looking” at it. 

I reverted to stack and only changed vtx_power to no avail. 

The osd shows something like ‘F:1:4’ when on power 4 but it’s hard to read.

BeauBrewski, 2022-04-10:
And just to verify, the Vtx works fine with BF, right?

simeonpashley, 2022-04-10:
Yeah, I drop back to BF4.3, stock config to keep flying without problems. Buzzing around my house.

Guessing it's firmware but the BF OSD shows `R:8:25` for the Race 8 25mW config, fuzzy Emu OSD shows `F:1:4` for stock with `vtx_power=4`. I know the band & channel are different. Power on BF shows mW where Emu shows enum. Just a different format?

I thought it was pitmode, but I've now tried Emu with a myriad of vtx via CLS & OSD, no changes in the VTX behaviour.
BF is fine, and OSD changes do affect the vtx.

Guessing it's different as the device openvtx also supports 400mW. I don't need this, but it doesn't seem to fit in the Emu enums.

BeauBrewski, 2022-04-10:
If OSD showed F:1:4, try changing your goggles to that setting (Band F, Channel 1, Freq 5740), as F:1 is default Emu settings. Its almost like the Vtx isn't taking the changes when set up in Emu CLI.

simeonpashley, 2022-04-10:
Bizarrely, seems to be working now after I just re-flashed Emu from the ZIP above?! Applied the `0803` pid preset, bind, modes, etc. No other changes, which is what I did last time.

Here's a CLI `get vtx`, it appears to be the same. Now wondering if my goggles were freaking out on my previous attempt, which would be doubtful. After my last Emu attempt, I reflashed BF4.3 and started flying.

I'll take it for a test flight but it all seems fine now and I can't put my finger on what was wrong.

```
# get vtx
vtx_band = 4
Allowed range: 0 - 5

vtx_channel = 1
Allowed range: 1 - 8

vtx_power = 1
Allowed range: 0 - 4

vtx_low_power_disarm = OFF
Allowed values: OFF, ON

vtx_freq = 5740
Allowed range: 0 - 5999

vtx_pit_mode_freq = 0
Allowed range: 0 - 5999

vtx_halfduplex = ON
Allowed values: OFF, ON
```

simeonpashley, 2022-04-10:
Nope - it's gone again. Trying to debug exact moment it stops. I'll update here.
After the vtx drops, I've done a channel scan in my goggles and nothing shows up.

BeauBrewski, 2022-04-10:
Got a screen cap of your PORTS tab? And a Dump All of your Emuflight install.

BeauBrewski, 2022-04-10:
Also, try switching to Tramp protocol instead of Smart Audio (or vice versa).

AYapejian, 2022-04-10:
I haven't tried the above firmware version, and if vtx works perfectly in BF then I suppose it's a moot point but I know I had to hot glue my vtx connector down as I was getting a lot off static, dropouts I couldn't nail down.  I might try the above linked fw if I have  some time today.  I'll update if I do.  ( This is for the mob6 v2 elrs which bf targets as `CRAZYBEEF4SX1280` just for reference )

BeauBrewski, 2022-04-10:
@AYapejian Emuflight does not currently support SPI ELRS boards that use the `CRAZYBEEFSX1280` target. Sorry.

@simeonpashley I'm seeing on the OpenVTx discord and other places where folks using Emuflight with OpenVTx were unable to get out of Pitmode when using SmartAudio, but everything worked when they switched to Tramp protocol. This might be worth looking into. https://github.com/OpenVTx/OpenVTx/issues/29

AYapejian, 2022-04-10:
@BeauBrewski gotcha, thanks.  I missed https://github.com/emuflight/EmuFlight/issues/612 which seems to cover what I'd need in emuflight ( if / when it's done ) thanks for clearing that up.

BeauBrewski, 2022-04-10:
@AYapejian hopefully we'll get the drivers ported before Emu 1.0 gets released 😆

simeonpashley, 2022-04-10:
> Got a screen cap of your PORTS tab? And a Dump All of your Emuflight install.

I reflashed back on the HEX above as I have been doing. Same process, I've been flipping between Emu & BF to try and get this working. BF works perfectly, so generally using that while debugging Emu.

I trued Race8, vtx_power=2 to no avail. I was hoping it would be 3rd time luck.

I posted screen grab of my ports on imgur, adjacent to BF ports - they're the same.
https://imgur.com/a/6FagrZN

`diff all`

```
diff all

# version
# EmuFlight / CRAZYBEEF4FR (C4FR) 0.4.0 Apr  9 2022 / 08:01:46 (7d9c91540) MSP API: 1.51

board_name 
manufacturer_id 
mcu_id 003f00313439511336383737
signature 

# reset configuration to default settings
defaults nosave

# name
name Mob6emu

# resources

# mixer

# servo

# servo mix

# feature
feature -AIRMODE

# beeper

# beacon

# map

# serial
serial 1 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# master
set fpv_mix_degrees = 90
set cinematic_yaw = ON
set dshot_idle_value = 550
set motor_pwm_protocol = DSHOT300
set yaw_motors_reversed = ON
set pid_process_denom = 2
set osd_vbat_pos = 2243
set osd_tim_1_pos = 2295
set osd_throttle_pos = 2263
set osd_crosshairs_pos = 2253
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 2
set vtx_freq = 5917
set frsky_spi_tx_id = 18,121
set frsky_spi_offset = -37
set frsky_spi_bind_hop_data = 5,120,1,115,230,110,225,105,221,100,215,95,210,91,205,85,200,80,195,75,190,70,185,65,180,60,175,55,170,50,165,45,160,40,155,35,150,30,145,25,140,20,135,15,130,10,125,0,0,0

# profile
profile 0

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

```

simeonpashley, 2022-04-10:
> @simeonpashley I'm seeing on the OpenVTx discord and other places where folks using Emuflight with OpenVTx were unable to get out of Pitmode when using SmartAudio, but everything worked when they switched to Tramp protocol. This might be worth looking into. [OpenVTx/OpenVTx#29](https://github.com/OpenVTx/OpenVTx/issues/29)

I tried swapping to Tramp - no difference. I went into the fuzzy OSD and changed vtx in there, no difference.

When running on USB, I started to smell something odd, the quad was crazy hot! The dome over the vtx was very very hot. I turned it off.

I appreciate running it on USB only doesn't give it cooling but I've never had this issue within just a minute or so of USB connection. BTW no battery, just USB for Emu.

BeauBrewski, 2022-04-10:
Okay, so I've had similar issue happen a lot when testing on the bench. Vtx will work great in the goggles then after half a minute everything goes away and can't see anything. Usually the VTx is over heating because no air flow. Its worse if you run higher power levels as well. Betafpv whoops are really bad about this because their canopy's have no ventilation at all. Mobula's aren't as bad, but my Mob6HD will get crazy hot, even just on USB power. 

A lot of folks have a fan on their bench to help prevent this. 

Another way to prevent this is to `set vtx_low_power_disarm = on` 
Your BF dump shows you had it turned on, but have it off on Emu which can be causing your Emu overheating. 
This will run your Vtx at lowest power setting until you arm and take flight to help prevent overheating of Vtx while battery plugged in, but quad not armed for flight with wind cooling the electronics. 

Another thing to help with heat is make sure your computer % is below 30%. Setting gyro/pid frequencies to 8k/4k or 4k/4k will help lower the processing % and keep things "cooler". 

[image]

simeonpashley, 2022-04-11:
@BeauBrewski Thanks for the reply. I had already tried `set vtx_low_power_disarm = on` to no avail, it seems like the power is super low anyway.
I had already tried 8k:2k loops, which were the BF defaults too. The CPU is very low and a CLI `tasks` supports the loops are accurate.

It's frustrating that no config/CLI/OSD changes appear to have any effect on the vtx. My attempts at debugging this haven't turned up any repeatable steps to make it work _or_ not work.

I would agree that it seems heat would play into the "it works", wait a while, "it's stopped working" aspect. It may not help but BF4.3 doesn't exhibit any of these issues, it remains at ambient temperature regardless of the bench USB time.

simeonpashley, 2022-04-11:
It may be worth closing this issue as the original ask was about arming, which has been solved. Unless this could be rephrased as a "Mobula6 v2" generic issue. I could raise another issue about VTX, which isn't related to arming. Happy to go with your guidance.

BeauBrewski, 2022-04-11:
Man, sorry we can't get the vtx issue figured out. That's frustrating especially since it was working ever so briefly. 
We can make a separate issue, put a reference link to this one. Close this one, and I'll push the gyro code forward. Maybe able to make some ground on what's going on with the openvtx Mob6 v2. 
I love Happy Models embracement of new features (ELRS, OpenVTX, etc), but it can be a pain too when stuff gets changed all the time.

closed 2022-04-11 by simeonpashley