emuflight/EmuFlight #796: Mobula6 v2 - OpenVTX output very low power, not responding to config changes

simeonpashley, 2022-04-11:
**Describe the bug**
Mobula6 v2 vtx fails

This was originally in issue #794 and pulled here to focus on the VTX issue as it remained unsolved despite help from @BeauBrewski 
https://github.com/emuflight/EmuFlight/issues/794#issuecomment-1094090839

It appears the vtx power is very low, I can see ghost shapes when I look at the quad when it's in my hand. Not enough to do anything with but I can see shapes if I move the quad+cam, e.g., looking at a bright window.

I did have 1 occurrence where the VTX worked clearly on the bench but it failed shortly afterwards and went back to 100% static.

I have tried various combinations of `vtx_*` settings including power and pit mode.

As per other thread, I have also tried swapping between SmartAudio and Tramp without results. https://github.com/OpenVTx/OpenVTx/issues/29

CLI `get vtx` output

```
# get vtx
vtx_band = 4
Allowed range: 0 - 5

vtx_channel = 1
Allowed range: 1 - 8

vtx_power = 1
Allowed range: 0 - 4

vtx_low_power_disarm = OFF
Allowed values: OFF, ON

vtx_freq = 5740
Allowed range: 0 - 5999

vtx_pit_mode_freq = 0
Allowed range: 0 - 5999

vtx_halfduplex = ON
Allowed values: OFF, ON
```

I've followed the Emuflight VTX guide to get this far,
https://emuflight.github.io/features/VTX-SmartAudio-and-Tramp.html#emuflight-osd

The [HappyModel website](https://www.happymodel.cn/index.php/2022/02/14/mobula6-v2-0-upgraded-flight-controller-crazybeef4fr-lite-crazybeef4fs-lite-v3-1-version-update-log/) implies a hardware change. Specifically the VTX has changed to OPENVTX

```
Mobula6 V2.0 upgraded flight controller CrazybeeF4FR lite / CrazybeeF4FS lite V3.1
The flight controller for Mobula6 and Mobula6 HD Frsky version and Flysky version would upgrade to V3.1 version since 14th .Feb 2022
Update log:
- MPU6000 changed to ICM20689 because of MPU6000 discontinued
- Onboard VTX changed to OPENVTX support 0mw~400mw
- Add “GND” solder pad for VTX antenna
- Only support Betaflight firmware 4.1.Xversion and latest than 4.1.X not include Legacy target.
```

**To Reproduce**
- Flash "new" Mobula6 Frsky with 0.4.0
- VTX is unusable through goggles.

**Expected behavior**
- Flash "new" Mobula6 Frsky with 0.4.0
- VTX is usable through goggles

**Flight controller configuration**
Stock EmuFlight 0.4.0 - no changes

**Setup / Versions**
Stock Mobula6 Frsky, purchase 3/Apr/2022
Happymodel website implies it's CrazybeeF4FR lite 

**Additional context**
Longer details on debugging in other issue https://github.com/emuflight/EmuFlight/issues/794#issuecomment-1094090839

BeauBrewski, 2022-04-11:
Out of curiosity and frustration, I ordered one of these to see if I can help diagnose and fix. Plus my other Mob6's are 19k motors and I wanted an excuse to get the 25k motors 😉 . Will update in about a week when it arrives, if problems aren't solved before then. 👍

BeauBrewski, 2022-04-14:
Okay, so I also have same symptoms of you when setting up to SmartAudio. Setting on high power, close to drone and can barely see anything. All changes had been saved from CLI and could be seen in OSD menu.
[image]

BeauBrewski, 2022-04-14:
Next, went back into configurator and set to TRAMP
[image]

Setup up VTx in CLI to make sure all my settings match as before and made sure to `save` any changes:
[image]

Plugged in battery and turned on goggles, and bammo, great picture:
[image]

simeonpashley, 2022-04-14:
It's interesting you seem to have set `vtx_power = 3` but the capture shows `A1 5865 25`, guessing the 25 is 25mW?

simeonpashley, 2022-04-14:
I captured a video through my goggles and got a screengrab below, on the bench via USB, no battery. Freshly flashed with the Mobule6v2 .hex.

- UART2 set to "IRC Tramp" as per your screenshot. 
- Changed VTX similar to yours but with Race8 (5:8)

```
# set vtx_band = 5
vtx_band set to 5
# set vtx_channel = 8
vtx_channel set to 8
# set vtx_power = 3
vtx_power set to 3
# set vtx_low_power_disarm = ON
vtx_low_power_disarm set to ON
# set vtx_freq = 5912
vtx_freq set to 5912
# set vtx_halfduplex = ON
vtx_halfduplex set to ON
# set vtx_pit_mode_freq = 0
vtx_pit_mode_freq set to 0
```

[image]

simeonpashley, 2022-04-14:
>On unrelated note, having issue with flight time on 3 out of 4 batteries that came with it. Not sure if bad charger or bad batteries.

I did wonder about this when the quad gets hot, it implies it's using a lot of power.

My loops are set to 8k:2k, CPU is ~25% on the bench but the quad is still hot. I would _guess_ that's heat coming from the vtx? I don't know enough to know if that's true or not.

BeauBrewski, 2022-04-14:
Processor and Vtx are going to get hot on these when on the bench, but I did not notice it being any worse than my other Mob6's. Battery drain was crazy fast on 3 out of 4 batteries. Like 5-10 second flights to get "Land Now" warning. It didn't leave a hot battery like there's a short or mechanical/electrical issue, so I'm thinking something else going on. Acts like saggy, old batteries.

nerdCopter, 2022-04-14:
@simeonpashley , FC's will get hot even when only using USB.  for "bench testing" always set to lowest VTX power, and also lowest usable pidloop.  use a fan when possible. I do a lot of EmuConfiogurator work and use unsoldered FC's to test. i always have to set pidloop to the LOWEST possible setting to keep from damaging hardware.  dynamic_filter requiores at least 4K loop, but for anything else i set 1K or even lower if the FC allows. For powered VTX, use a fan to keep it cool even at lowest mW power.

simeonpashley, 2022-04-14:
@BeauBrewski 
> Battery drain was crazy fast on 3 out of 4 batteries. Like 5-10 second flights to get "Land Now" warning

After a quick test flight, I changed the config to the `0802` PID preset and some elements I use on my Mob7 from [PMB](https://projectmockingbird.squarespace.com/pmb-setup-guides/pmb-brushless-emu), flight times seem to be on par with BF4.3.

With respect to the original Mobula6 v2 "issue", using the setting for "IRC Tramp" on UART2, _and_ maybe the vtx_power=3(?), seems to be solvable by config.

BeauBrewski, 2022-04-14:
@simeonpashley 
We added a note to the VTX wiki that mentions setting OpentVTx transmitters to IRC Tramp to prevent locked in pit mode issue. This was the fix I had found from searching OpenVTx discord and github issues. 

Thanks for bringing to our attention the issue of the Vtx and the gyro change on new Mobulas. That was a much easier fix than getting SPI ELRS to work 😆 

https://github.com/emuflight/EmuFlight/wiki/VTX-SmartAudio-and-Tramp

simeonpashley, 2022-04-14:
@BeauBrewski 
> Battery drain was crazy fast on 3 out of 4 batteries. Like 5-10 second flights to get "Land Now" warning

Confirmed, a flow about 10m, batt dropped to 2.9v and I dropped out of the sky. Roughly 10s of flight time.

Sadly, back to BF4.3 :/

simeonpashley, 2022-04-14:
Re-flash BF4.3 - thought it worth sharing that, when on the bench USB for a few minutes, the quad doesn't get super hot like it did with Emu, it is warm.

BeauBrewski, 2022-04-15:
Crap. Flashed to BF and setup and got 3:42 flight time just hovering and gently flying around my room. 
Good news is that my batteries are good. Bad news is that the Emuflight target has some battery issue.

BeauBrewski, 2022-04-15:
@simeonpashley 
Getting longer flight times on Emu now.
My first step that didn't give results, but gave better readings of amperage time was to set mah to 300mah on the battery tab and set the Amperage scale to what HM has listed as their default for the V2 (only really matters in amperage reading on OSD, but if voltage is being goofy, figured it wouldn't hurt).
[image]

 I remember there was an issue on PMB and one way to fix it was to turn on `Dynamic Filter` on the Configuration Tab
[image]

And at same time I noticed that BF default Motor Poles is 12, and Emu had 14, so I went ahead and changed that.
```
# get motor_poles
motor_poles = 14
Allowed range: 4 - 255
Default value: 14

set motor_poles = 12
motor_poles set to 12
save
```
And out of desperation, I went ahead and set the battery cell count to 1
```
force_battery_cell_count = 0
Allowed range: 0 - 24
Default value: 0

set force_battery_cell_count = 1
force_battery_cell_count set to 1
save
```
And I flew 2 packs in a row that lasted over 2 minutes.

BeauBrewski, 2022-04-19:
@simeonpashley any of those fixes work for you? I haven't had time to get some batteries charged and really get a good tune setup yet.

BeauBrewski, 2022-04-22:
Finally getting batteries put through it, getting easy 3:30 minute flight times, and IRC Tramp protocol working great with the OpenVTX. Closing issue as Vtx issue solved. 

[image]

closed 2022-04-22 by BeauBrewski

simeonpashley, 2022-04-22:
Hi @BeauBrewski, I've been away for a few days and just got back. It looks like you've solved it and I'll definitely be trying it out

simeonpashley, 2022-04-22:
It all appears to be working just fine! I got 4:30 just hovering around my desk, that was previously <20s before it fell out of the sky.

For future googlers, at the time of writing this uses a custom .hex file for the Mobula6v2 and I believe these are the relevant settings to fix VTX & battery life
```
# serial - UART2=IRC Tramp
serial 1 8192 115200 57600 0 115200
# vtx fix
set vtx_power = 3
set vtx_low_power_disarm = ON
# motor poles for DSHOT
set motor_poles = 12
# battery
set bat_capacity = 300
set force_battery_cell_count = 1
set ibata_scale = 1175
```

Here's my `diff all` with my tweaks for receiver, OSD, rates and PIDs (from the 0802 pid preset).

```
diff all

# version
# EmuFlight / CRAZYBEEF4FR (C4FR) 0.4.0 Apr  9 2022 / 08:01:46 (7d9c91540) MSP API: 1.51

board_name 
manufacturer_id 
mcu_id 003f00313439511336383737
signature 

# reset configuration to default settings
defaults nosave

# name
name mob6emu

# resources

# mixer

# servo

# servo mix

# feature
feature -TELEMETRY
feature -AIRMODE

# beeper

# beacon
beacon RX_LOST

# map

# serial
serial 1 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 2 1700 2100 0 0
aux 2 2 2 1300 1700 0 0
aux 3 28 2 900 1700 0 0
aux 4 35 3 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set imuf_roll_q = 7000
set imuf_pitch_q = 7000
set imuf_yaw_q = 7000
set imuf_w = 8
set dynamic_gyro_notch_q = 450
set fpv_mix_degrees = 90
set cinematic_yaw = ON
set dshot_idle_value = 550
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set bat_capacity = 300
set vbat_max_cell_voltage = 45
set vbat_min_cell_voltage = 29
set vbat_warning_cell_voltage = 30
set force_battery_cell_count = 1
set ibata_scale = 1175
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set pid_process_denom = 4
set osd_vbat_pos = 2241
set osd_tim_1_pos = 2328
set osd_flymode_pos = 2424
set osd_throttle_pos = 2298
set osd_crosshairs_pos = 2253
set osd_mah_drawn_pos = 2273
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5917
set vcd_video_system = NTSC
set frsky_spi_tx_id = 18,121
set frsky_spi_offset = -43
set frsky_spi_bind_hop_data = 5,120,1,115,230,110,225,105,221,100,215,95,210,91,205,85,200,80,195,75,190,70,185,65,180,60,175,55,170,50,165,45,160,40,155,35,150,30,145,25,140,20,135,15,130,10,125,0,0,0

# profile
profile 0

set throttle_boost = 20
set i_decay = 6
set emu_boost = 500
set emu_boost_yaw = 500
set emu_boost_limit = 50
set emu_boost_limit_yaw = 50
set p_pitch = 76
set i_pitch = 80
set d_pitch = 40
set p_roll = 70
set i_roll = 80
set d_roll = 40
set p_yaw = 75
set i_yaw = 80
set d_yaw = 10
set mixer_laziness = ON

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 1
set pitch_rc_rate = 1
set yaw_rc_rate = 210
set roll_srate = 0
set pitch_srate = 0
set yaw_srate = 0
set tpa_rate_p = 100
set tpa_rate_i = 165
set tpa_rate_d = 95
set tpa_breakpoint = 1400

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
# 
```