emuflight/EmuFlight #800: Target Request: MAMBAF722_X8

krslim, 2022-04-26:
**Vendor/Model**
MAMBAF722_X8
Diatone MAMBA EASY F722 MK1 Flight Controller
https://www.diatone.us/products/mamba-easy-f722-mk1-55a-stack?_pos=1&_sid=a8ca3b484&_ss=r
Note:  MAMBAF722 target works (quadcopter layout) but without barometer.

**CLI dump or Unified Target**
https://github.com/betaflight/unified-targets/blob/master/configs/default/DIAT-MAMBAF722_X8.config

BeauBrewski, 2022-04-27:
Should have you a test target by the end of the week. 👍

krslim, 2022-04-27:
Thanks!
Barometer isn't showing after pasting the CLI commands into `MAMBAF722` target, I double checked resources after & they were assigned as intended.

But the `MAMBAF722I2C` target does work & baro is now there, getting a good reading in the sensors tab.  I've not test flown it yet but it looks good.

krslim, 2022-04-27:
Not the best pic, possibly BMP280?  It measures correctly (2.5 x 2.0mm)
[image]

BeauBrewski, 2022-04-28:
@krslim awesome. So we have correct driver, just the regular MambaF722 isn't setup for it, but I2C version is. I'll still make a target, since MAMBAF722I2C doesn't have all 8 motor assignments like MAMBAF722_X8 should.

krslim, 2022-04-28:
@BeauBrewski Thanks again.  The standard target otherwise worked fine so I will try out MAMBAF722I2C for now & let you know if there's any issues.

BeauBrewski, 2022-04-28:
Needs tested!
Check gyro orientation, motor arrangement, flash, osd, vtx control, everything else if possible please update here with feedback that it works or needs improvement.
PROPS OFF FOR SAFETY

[EmuFlight_0.4.0_MAMBAF722_X8_Build_bc1e90621.zip](https://github.com/emuflight/EmuFlight/files/8585339/EmuFlight_0.4.0_MAMBAF722_X8_Build_bc1e90621.zip)

krslim, 2022-04-28:
Awesome!  Will flash it & let you know how it goes.

krslim, 2022-04-28:
No barometer again this time.  Everything else works fine on the bench but it does takes a little longer for the board to boot than usual.

Would a dump from the BF 4.3 target be helpful?  Or from any of these targets?

BeauBrewski, 2022-04-28:
Let me double check some things, I think I just have some definitions crossed up.

BeauBrewski, 2022-04-28:
@krslim 
Okay, I updated the zip file above. There was a resource assignment conflict I fixed and baro should be showing up now (tested on a similar F722 fc).

krslim, 2022-04-28:
It looks good :)

Will give it a more thorough test tomorrow & report back.

krslim, 2022-05-03:
I've not come across any issues with this target & my setup, it's working as the `MAMBAF722` target did but with baro active.  
The barometer does drift around quite a bit but that seems like a common issue in general!

BeauBrewski, 2022-05-05:
Merged. Closing issue.

closed 2022-05-05 by BeauBrewski