emuflight/EmuFlight #803: [Firmware Bug Report] MAMBAF722/MAMBAF722I2C gyro MPU6500 addition

BeauBrewski, 2022-04-28:
Gyro changes to Diatone Mamba MK2  FC have caused current MPU6000 setup on targets `MAMBAF722/MAMBAF722I2C` not to work.

Contact with Diatone has returned with the gyro change being MPU6500.

https://www.diatone.us/collections/dji/products/mamba-dji-f722-mk2-50a-6s-8bit-flight-controller-stack

Will add additional definitions for the MPU6500 gyro.

nerdCopter, 2022-04-28:
jeez, hardware modifications causing complications.
Thanks @BeauBrewski !

BeauBrewski, 2022-04-28:
Needs tested!
Please check gyro function, board orientation, motors, flight, etc.

[EmuFlight_0.4.0_MAMBAF722I2C_Build_bc1e90621.zip](https://github.com/emuflight/EmuFlight/files/8585309/EmuFlight_0.4.0_MAMBAF722I2C_Build_bc1e90621.zip)

BeauBrewski, 2022-05-04:
Tested by @Noodles on the discord page and flew great. Mergible if approved.

noodles, 2022-05-04:
@BeauBrewski - you've got the wrong noodles. I have no idea what this project is - looks cool though :)

BeauBrewski, 2022-05-05:
Hahahah, Oops, didn't mean to tag you. Sorry about that!

BeauBrewski, 2022-05-05:
Closing, Merged.

closed 2022-05-05 by BeauBrewski