emuflight/EmuFlight #805: Target request HAKRCF411D

torju01, 2022-05-04:
Vendor and model:  HAKRC F411 AIO 20A (F4126)
target name:  HAKRCF411D

vendor link:  http://www.hikrc.com/pd.jsp?id=63#_pp=103_435

unified targets:  https://github.com/betaflight/unified-targets/blob/master/configs/default/HARC-HAKRCF411D.config

BeauBrewski, 2022-05-05:
I actually have this exact FC and it came with `MATEKF411` flashed.  Flies great on Emuflight with that target.

BeauBrewski, 2022-05-05:
Awesome! I've had good luck with that FC so far. Other sellers websites also recommend `MATEKF411` for it.

https://pyrodrone.com/products/hakrc-f4126-blheli_s-20a-whoop-aio-fc

BeauBrewski, 2022-05-06:
Closing for now as most sites recommend `MATEKF411` for this board, and my own personal FC came with `MATEKF411` flashed for BF, and op has tested and flown with no problems.

closed 2022-05-06 by BeauBrewski